vshymanskyy/TinyGSM #99: A6 GPRS CONNECTION

tonyputi, 2017-11-05:
Hi. I'm testing A6 module (IoT GA6), all seem to work well but not the gprsConnect function.
As you can see in document [A6 A7 A6C A20 AT instructions v1.03.pdf](https://github.com/vshymanskyy/TinyGSM/blob/master/extras/doc/A6%20A7%20A6C%20A20%20-%20AT%20instructions%20v1.03.pdf
) hosted by your project there not is an AT commad called **CIPMUX=?**

bool gprsConnect(const char* apn, const char* user, const char* pwd) {
    gprsDisconnect();

    sendAT(GF("+CGATT=1"));
    if (waitResponse(60000L) != 1)
      return false;

    // TODO: wait AT+CGATT?

    sendAT(GF("+CGDCONT=1,\"IP\",\""), apn, '"');
    waitResponse();

    if (!user) user = "";
    if (!pwd)  pwd = "";
    sendAT(GF("+CSTT=\""), apn, GF("\",\""), user, GF("\",\""), pwd, GF("\""));
    if (waitResponse(60000L) != 1) {
      return false;
    }

    sendAT(GF("+CGACT=1,1"));
    waitResponse(60000L);

    sendAT(GF("+CIPMUX=1"));
    if (waitResponse() != 1) {
      return false;
    }

    return true;
  }

I also searched from ai-thinker directly and I've found the same document without any command **CIPMUX=?**

I've also tested gprs feature directy via AT command without any MCU and it seem to work. I'm able to maka an http request and get the content.

Could you please help me to understand how to get it worked?

Rgds

tonyputi, 2017-11-06:
and again modemConnect function wait for a **+CIPNUM:** response that seem not exists on official AI-Thinker AT manual:

```
bool modemConnect(const char* host, uint16_t port, uint8_t* mux) {
    sendAT(GF("+CIPSTART="),  GF("\"TCP"), GF("\",\""), host, GF("\","), port);

    if (waitResponse(75000L, GF(GSM_NL "+CIPNUM:")) != 1) {
      return false;
    }
    int newMux = stream.readStringUntil('\n').toInt();

    int rsp = waitResponse(75000L,
                           GF("CONNECT OK" GSM_NL),
                           GF("CONNECT FAIL" GSM_NL),
                           GF("ALREADY CONNECT" GSM_NL));
    if (waitResponse() != 1) {
      return false;
    }
    *mux = newMux;

    return (1 == rsp);
  }
```

another thing is about the mux variabe. In AT manual (page 23) there is a command:
AT+CMUX=<transparency> that could be only 0.

Again in this function CIPSEND expect mux and lenght  but in AT manual (page 156) seem that CIPSEND don't need any arguments.

```
int modemSend(const void* buff, size_t len, uint8_t mux) {
    sendAT(GF("+CIPSEND="), mux, ',', len);
    if (waitResponse(2000L, GF(GSM_NL ">")) != 1) {
      return -1;
    }
    stream.write((uint8_t*)buff, len);
    stream.flush();
    if (waitResponse(10000L, GFP(GSM_OK), GF(GSM_NL "FAIL")) != 1) {
      return -1;
    }
    return len;
  }
```
Could someone help me?

mixaz, 2017-11-12:
On my A7 CIPMUX and CIPNUM seem to work OK:

```
AT+CIPMUX=1
### ..: OK , ERROR ,  ,  , 
OK
 OK
Connecting to cdn.rawgit.comAT+CIPSTART="TCP","cdn.rawgit.com",80
### ..: +CIPNUM: , ERROR ,  ,  , 
+CIPNUM:0
### ..: CONNECT OK , CONNECT FAIL , ALREADY CONNECT ,  , 
CONNECT OK
```
I believe CIPMUX was added since some firmware revison, so you probably have an old A6

tonyputi, 2017-11-12:
My module show IoT-GA6 instead of AI A6. probably it is a clone.

Could someone confirm this?

Rgds

mixaz, 2017-11-12:
Not sure how someone can confirm it.  What firmware does the module report on ATI? Mine is 
```
Ai Thinker Co.LTD
A7 
V03.03.20161229019H03
```

tonyputi, 2017-11-12:
Mine this:

Kingcom_Ltd
Kingcom_G16
Revision: G16_B5_R04_A02_D161111

so I think that is a clone

jishnu-jdn, 2018-01-23:
I have an A6 with "Revision: G16_B5_R05_GUOYUN_A01_D170727"
it returns "+CME ERROR:53" on CIPMUX.

Black3rror, 2018-02-18:
Hello.I have a IOT-GA6 like what u have and when I send at+cipmux=? with AT_DEBUG example I'll get  

> +CIPMUX: (0,1)
> 	
> OK

so it works good for me.also I have another problem with this module.
first: module can't work correctly with voltage less than 5v. but in its datasheet its been written that its voltage range is 3.5 to 4.2 volt.so why it works like this?
second: on voltage 5.2v it gets commands and sends responses. but it works in such a bad way. as u can see in the pictures below sometimes it gets command and sends response with a bad quality and it ends up to failure. surprise is that when I use Serial-monitor and USB-TTL for communicating with A6, it works nice but using arduino or NodeMCU causes problem. so can someone tell me whats problem?
[image]
[image]

and its power supply is in this way : 2-Ampere 6-volt adapter to lm2596 module which makes a 5.2v for A6mini. the picture below shows power supplying : 
[image]

hubaksis, 2018-02-18:
@Black3rror 
Firstly, you need to fix the power.
Solderess board's connection won't give you proper 2A. I even had issues with a constant 0.5A. So, either solder it or connect a dupont cable from the power module directly to the modem.

3.5 to 4.2V is the voltage for the chip (not the module). You are working with the module, so the voltage is different. 5V usually fine but check your module description.

Secondly - connection.
SoftwareSerial has issues working on high speed. Try to change the module speed when you are connected directly to the module and then connect your arduino.

sribence, 2018-03-02:
@Black3rror The problem will be with the fast baud rate. I've got the same result as you with the 115200 baud, but after setting it to 9600, it works for me !
But im also get +CME ERROR:53 if I try the AT+CMUX=1 command. Anybody got solution for this?

sribence, 2018-03-02:
@jishnu-jdn @tonyputi I think I managed to solve the problem with a firmware update (A6-V03.06.20171127R).
So the kingcom modules needs an update, as described here: http://wiki.ai-thinker.com/gprs_download.

ryansuchocki, 2018-03-03:
Can you elaborate on the firmware upgrade procedure? I downloaded the 'download tool' and firmware from the ai-thinker wiki, but the download tool doesn't seem to do anything.

uudecode, 2018-03-03:
@ryansuchocki  what the problem? did you connect AI-thinker module not for standard TX/RX but other, "host" TX/RX ?
I upgraded few modules, it easy. Explain your problem please.

uudecode, 2018-03-03:
**@here guyz, important tip for A6!**
I check few A6 modules, after first tcp-connect and get first tcp-data A6 just reboot.
I found workaround:
`// MQTT STUFF
void mqttConnect() {
  MQTTOn = false;
  if ((millis() - lastReconnectMQTT ) <  reconnectMQTTTimeout && lastReconnectMQTT > -1) {
    myGLCD.setColor(MATERIAL_PURPLE);
    myGLCD.fillRect(0, 0, 319, 13);
    return;
  }
  lastReconnectMQTT = millis();
  myGLCD.setBackColor(MATERIAL_PURPLE);
  myGLCD.setColor(MATERIAL_WHITE);
  mqtt.disconnect();
  Serial.println("Connecting to MQTT");
  myGLCD.print("Connecting to MQTT", TOP_STRING);
  client.connect(broker, 1883);  /* Cuz  A6 bad behavior of my copies opf A6 we have to open*/
  client.stop(); /* and  close TCP-connection */

  MQTTOn = mqtt.connect(mqttClientID,mqttUser,mqttPassword);
  myGLCD.setColor(MATERIAL_PURPLE);
  myGLCD.fillRect(0, 0, 319, 13);
  printStatusOnTFT();
  if (MQTTOn) { 
    mqtt.subscribe(topicRequest);
    mqtt.subscribe(topicServerHearthBeat);
  }
  myGLCD.setColor(MATERIAL_PURPLE);
  myGLCD.fillRect(0, 0, 319, 13);
  myGLCD.setBackColor(MATERIAL_PURPLE);
  myGLCD.setColor(MATERIAL_WHITE);
  myGLCD.print("Touch active", TOP_STRING);

}
`

IMAN4K, 2018-03-21:
Hi @uudecode 
Did you managed to successfully burn the image to A6 ?
I went through the guide from here but after about 2 sec of USB adapter(`CH340G`) connection we get unconnected state!

[image]
Could you help us on this. 

Regards.
IMAN.

vshymanskyy, 2018-03-24:
IoT-GA6 seems to be not the same as AI A6, and it is not supported/tested.
Closing for now.

closed 2018-03-24 by vshymanskyy

vshymanskyy, 2018-03-24:
It is now documented in the README: https://github.com/vshymanskyy/TinyGSM/blob/master/README.md#goouuu-tech-iot-ga6-vs-aithinker-a6-confusion

emigbur, 2018-09-02:
Hi I got A6 Module  Revision: GA6-B_B5_R01_A01_D180428
Its works fine but im not able to connect to my web

AT+CIPSTART="TCP","acuariosiotchart.esy.es/",80
OK
STATE:IP PROCESSING
CONNECT FAIL

Whit this url are fine 
AT+CIPSTART="TCP","mujkotel.wz.cz",80 
OK
CONNECT OK

how can this been possible?
Please help.

Whit my other A6 Module Kingcom_Ltd
Kingcom_G16
Revision: RDA BV5

Kingcom_G16
Revision: G16_B5_R05_GUOYUN_A01_D170727
I can connect whitout any prolem, how can this be possible.?

regards

vshymanskyy, 2018-09-03:
Did you read the README?

wacanam, 2018-09-15:
Hello, i got my new IOT GA6,
And my problem is it keep on sending 

AT ready  

AST POWERON

+CREG: 3

repeatedly. What wrong with this?

Deewansonic, 2023-10-26:
@wacanam and to others that may later have this kind of problem. I have gotten solution to this problem. 
My problem was:

AT Ready

AST_POWERON

+CREG: 2

 I connected a capacitor across the IOT-GA6B's VCC and GND terminal and it worked. Capacitor specification is 1000uF by 25V.

wacanam, 2023-10-26:
Also, add 100nf mylar cap in parallel.