vshymanskyy/TinyGSM #102: I have problems with SAMD21 (Feather M0) and SIM808

RJakub, 2017-11-12:
Hi

I am using a Feather M0 Basic Board (SAMD21) with the sim808 module. 
Sim808 is connected via UART with the Serial1 port of the Feather Board.

**Problem**: 
I get no data or bad data when I try to use the WebClient sketch from the examples. 

I tried the same sketch with an Arduino Mega (ATMega 2560) without any problems.
I also tried different versions of the TinyGSM library. But it didn't help. Currently using the latest one.  

**Code**:

```
#define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_ESP8266

#include <TinyGsmClient.h>

// Your GPRS credentials
// Leave empty, if missing user or pass
const char apn[]  = "web.vodafone.de";
const char user[] = "";
const char pass[] = "";

// Use Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial1

// or Software Serial on Uno, Nano
//#include <SoftwareSerial.h>
//SoftwareSerial SerialAT(2, 3); // RX, TX

TinyGsm modem(SerialAT);
TinyGsmClient client(modem);

const char server[] = "cdn.rawgit.com";
const char resource[] = "/vshymanskyy/tinygsm/master/extras/logo.txt";

const int port = 80;

void setup() {
  // Set console baud rate
  Serial.begin(9600);
  delay(10);

  // Set GSM module baud rate
  SerialAT.begin(9600);
  delay(3000);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  Serial.println(F("Initializing modem..."));
  //modem.init();

  String modemInfo = modem.getModemInfo();
  Serial.print("Modem: ");
  Serial.println(modemInfo);

  // Unlock your SIM card with a PIN
  //modem.simUnlock("1234");
}

void loop() {
  Serial.print(F("Waiting for network..."));
  if (!modem.waitForNetwork()) {
    Serial.println(" fail");
    delay(10000);
    return;
  }
  Serial.println(" OK");

  Serial.print(F("Connecting to "));
  Serial.print(apn);
  if (!modem.gprsConnect(apn, user, pass)) {
    Serial.println(" fail");
    delay(10000);
    return;
  }
  Serial.println(" OK");

  Serial.print(F("Connecting to "));
  Serial.print(server);
  if (!client.connect(server, port)) {
    Serial.println(" fail");
    delay(10000);
    return;
  }
  Serial.println(" OK");

  // Make a HTTP GET request:
  client.print(String("GET ") + resource + " HTTP/1.0\r\n");
  client.print(String("Host: ") + server + "\r\n");
  client.print("Connection: close\r\n\r\n");

  unsigned long timeout = millis();
  while (client.connected() && millis() - timeout < 10000L) {
    // Print available data
    while (client.available()) {
      char c = client.read();
      Serial.print(c);
      timeout = millis();
    }
  }
  Serial.println();

  client.stop();
  Serial.println("Server disconnected");

  modem.gprsDisconnect();
  Serial.println("GPRS disconnected");

  // Do nothing forevermore
  while (true) {
    delay(1000);
  }
}
```

**Serial monitor output**: 

```
//First Try

Initializing modem...
Modem: SIM808 R14.18
Waiting for network... OK
Connecting to web.vodafone.de OK
Connecting to cdn.rawgit.com OK
HTTP/1.1 200 OK
Date: Sun, 12 Nov 2017 14:41:53 GMT
Content-Type: text/plain;charset=utf-8
X-Content-Type-Options: nosniff
X-Robots-Tag: none
Access-Control-Allow-Origin: *
ETag: "0d8bbf390498c7af25cf8862b358100a667c8919"
Cache-Control: max-age=86400
Vary: Accept-Encoding
RawGit-Cache-Status: HIT
Server: NetDNA-cache/2.2
X-Ca⸮he: *⸮⸮~Con⸮Y,W⸮⸮K⸮⸮͕5

  _____ 
           _____  _____  _____
    |  ⸮ |\ ⸮\_⸮ | ⸮⸮⸮⸮}}}}}⸮

// Second Try

Initializing modem...
Modem: SIM808 R14.18
Waiting for network... OK
Connecting to web.vodafone.de OK
Connecting to cdn.rawgit.com fail
Waiting for network... OK
Connecting to web.vodafone.de OK
Connecting to cdn.rawgit.com fail
Waiting for network... OK
Connecting to web.vodafone.de OK
Connecting to cdn.rawgit.com OK

Server disconnected
GPRS disconnected
```

RJakub, 2017-11-19:
I could solve the problem. 
I didn't use the hardware serial port (D0, D1) but a sercom uart port (e.g. D10, D12). 
To configure the sercom port for uart, I used the following code:

```
#define PIN_SERIAL2_RX       (34ul)               // Pin description number for PIO_SERCOM on D12
#define PIN_SERIAL2_TX       (36ul)               // Pin description number for PIO_SERCOM on D10
#define PAD_SERIAL2_TX       (UART_TX_PAD_2)      // SERCOM pad 2
#define PAD_SERIAL2_RX       (SERCOM_RX_PAD_3)    // SERCOM pad 3

Uart Serial2(&sercom1, PIN_SERIAL2_RX, PIN_SERIAL2_TX, PAD_SERIAL2_RX, PAD_SERIAL2_TX);
```

I do not know why the hardware port is not working correctly, but with a sercom port everything is working perfectly.

vshymanskyy, 2017-12-14:
Thanks for the report!

closed 2018-02-09 by vshymanskyy

nielbierman, 2018-08-15:
I am having the same problem using a different board. I use the [Moteino M0](https://lowpowerlab.com/2018/05/27/moteino-m0-released/) form LowPowerLabs. It is based on the Zero I think. Unfortunately the code above does nor work for me, or maybe I'm not using the correct pins? Not sure how the pin mapping differs from the Zero. Also not sure what the (34ul) mean.
I hope someone can give me some idea on how to get sercom working on my board.

commit dc1506c9 (2018-02-09): Update readme. Fix #102