vshymanskyy/TinyGSM #104: ADD ESP8266 Examples

ecuellar, 2017-11-22:
Excelent lib.  Is there any examples for the ESP8266? or documentation?

Regards

daviidmart, 2017-11-22:
All the examples are for the ESP8266

ecuellar, 2017-11-23:
thanks for your answer, but for example: I tried the diagnostics example:

// Select your modem:
// #define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
  #define TINY_GSM_MODEM_ESP8266

I uncomment the last line, and now when I tried to compile, its tells me that there is no method called"modem.gprsConnect()".. so I open some libraries, and found there is a method called "modem.networkConnect()".. I changed, it compiles, but the example does not run very well, does not connect to the network.

Br4inslug, 2018-01-16:
Hey Guys i also think its a great lib.
But im running in the same problems like ecuellar with esp8266 the example and the diagnostic runs but i was not able to connect to a network.

Regards

vshymanskyy, 2018-02-03:
ESP8266 is experimental, and I actually wouldn't recommend using it. I'm not providing examples, as WiFi modules require different setup, etc. And partially because I don't want to maintain it at the moment.. Maybe when developer community of this project grows a little bit, or at some point later guys ;)

closed 2018-02-03 by vshymanskyy

ClemensGruber, 2021-05-10:
I'm using TinyGSM on the TTGO T-Call, an ESP32 with SIM800 module. So my question would be can I use the build in WiFi module of the ESP32 with TinyGSM to submit alternatively data via Wifi? So is the ESPxx example meant for an ESP as microcontroller with built in Wifi or is it for any microcontroller with an "external" ESP connected to the microcontroller as modem?

SRGDamia1, 2021-05-11:
@ClemensGruber The ESPxx code in this library is intended only for a microcontroller with an "external" ESP.  That external ESP must be running Espressif's "AT" firmware.

ClemensGruber, 2021-05-12:
Many thanks for the clarification Sara, I've hoped for a convenient software solution with an easy switch to use e.g. HTTP or MQTT with the same lib and switcheing easily from GPRS to WiFi but using the same lib and underlaying concepts on an ESPxx. Ok so it must be implemented twice.