vshymanskyy/TinyGSM #105: Support for SIM7XXX LTE modules!

tusharvb19, 2017-11-23:
Can we port the existing library to work with the LTE modules?
Thanks.

tusharvb19, 2017-11-24:
Can we modify TinyGSM like this where the main adafruit library is modified!
https://github.com/botletics/NB-IoT-Shield

vshymanskyy, 2017-11-25:
Yes you can. Pull Requests are welcome!

tusharvb19, 2017-11-28:
I am beginner can anyone modify it and test it?
Thank you.

vshymanskyy, 2017-12-14:
If someone is willing to donate a board, I can take a look at it..

anthonyvage, 2018-09-26:
Let me know if i can send you a SIM7000 module

finchamp, 2018-10-04:
I would also like to see support for SIM7000/C/E. If someone can get him a module, I'll put a bounty on it.

finchamp, 2018-10-04:
This guy claims to have Hologram working with SIM7000?: https://github.com/botletics/SIM7000-LTE-Shield

jhonoryza, 2018-11-19:
please add this support guys 👍 :)

botletics, 2018-12-13:
I can send him a free shield kit.

SRGDamia1, 2019-05-08:
Support for the SIM7000 was added courtesy of https://github.com/captFuture.

closed 2019-05-08 by SRGDamia1

anatom74, 2019-05-09:
This is only support for SIM7000. What about SIM75XX and SIM76XX?