vshymanskyy/TinyGSM #114: 'TinyGsm' has no member named 'isGprsConnected'

lathoub, 2017-12-17:
Hi

My code:
`
#define TINY_GSM_MODEM_SIM800
`
...
`
#include <TinyGsmClient.h>
`
`
TinyGsm modem(SerialAT);
`
...
`
 if (!modem.isGprsConnected())
  {
    modem.gprsConnect(apn, user, pass);
  }
`

I get the following compile error:

`
ProjectTiny:231: error: 'TinyGsm' has no member named 'isGprsConnected'
`

Yet `isGprsConnected` is defined in `TinyGsmClientSIM800.h`
Can someone see what I'm doing wrong?

Thanks

closed 2017-12-26 by lathoub

vossie123, 2020-07-23:
I am having the same issue
'class TinyGsm' has no member named 'isGprsConnected'
also
'class TinyGsm' has no member named 'localIP'

SRGDamia1, 2020-07-23:
I'm sorry @vossie123, I can't help you much without more detail.  Please open a new issues instead of commenting on a fairly old closed one.