vshymanskyy/TinyGSM #115: trying to implement A6 read sms

axwell, 2017-12-18:
Hello,
i'm trying to implement sms reading in A6 by reading reading SMS locations of unread sms with command "+CMGL=REC UNREAD"

Here is my code:
        sendAT(GF("+CSCS=UCS2"));
        waitResponse();

        sendAT(GF("+CMGF=1"));
        waitResponse();

        sendAT(GF("+CNMI=1,0"));
        waitResponse();

        sendAT(GF("+CMGL="), GF("REC"), GF(" "), GF("UNREAD"));
        if(waitResponse(GF(GSM_NL "+CMGL:"), GF("OK"))) {
                data = stream.readString();
                Serial.print("message: ");
                Serial.println(data);
        }

The issue is that i receive only garbage data and i don't know why.
Library works just fine otherwise.

Example of received data:

[image]

Can someone help me ? I'm stuck.

shivsey, 2018-01-13:
Did you manage to read a valid SMS?  Im interested in doing the same with an A7,

axwell, 2018-01-13:
@shivsey yes
Add this two methods.

```
int getUnreadSMSLocs(int* buf)
{
        String data;

        int index, occurrences = 0;

        sendAT(GF("+CMGL=\"REC UNREAD\""));
        if(waitResponse(5000L, GF("+CMGL:"), GFP(GSM_OK), GFP(GSM_ERROR))) {
                do {
                        data = stream.readStringUntil('\n');
                        int pos = data.indexOf("REC UNREAD");
                        if (pos > 0) {
                                index = data.substring(0, pos - 1).toInt();
                                buf[occurrences] = index;
                                occurrences++;
                        }
                } while (waitResponse() == 1);
        }
        return occurrences;
}

SMSmessage readSMS(uint8_t i)
{
        char message[300];
        char number[50];
        char date[50];
        char type[10];

        String head;
        String body;

        SMSmessage sms = (const struct SMSmessage) {
                "", "", ""
        };
        sendAT(GF("+CMGR="), i);

        if (waitResponse(10000L, GF(GSM_NL "+CMGR:"))) {

                head = stream.readStringUntil('\n');
                body = stream.readStringUntil('\n');

                sscanf(head.c_str(), "\"REC %s\",\"%s\",,\"%s\"\r\n", type, number, date);
                sms.number  = String(number);
                sms.date    = String(date);

                body.toCharArray(message, body.length());
                sms.message = message;
        }
        return sms;
}
```

Add struct at the top next to enum things

```
struct SMSmessage {
        String number;
        String date;
        String message;
};
```

Red messages like this in the loop 
```
unreadSMSNum = modem.getUnreadSMSLocs(unreadSMSLocs);

                for (int i = 0; i < unreadSMSNum; i++) {
                        Serial.print("New message at index: ");
                        // Serial.println(unreadSMSLocs[i], DEC);

                        sms = modem.readSMS(i);
                }
```

Add method:

```
void prefSettings()
{
        sendAT(GF("+CPMS=ME,ME,ME"));
        waitResponse();

        sendAT(GF("+CMGF=1"));
        waitResponse();

        sendAT(GF("+CNMI=1,0"));
        waitResponse();

        sendAT(GF("+CSCS=\"GSM\""));
        waitResponse();
}
```

Initialize it after you have a valid operator.
Initialize serial connection with a bigger buffer, like 512

I hope you understand i did not have time to make a merge request.

shivsey, 2018-01-13:
Thanks, got it to compile no problem, still waiting to receive the module but the code looks pretty straight forward.  If you have anymore A6/A7 specific code let me know.

jeancaffou, 2019-07-06:
> First, I modified the structs. I use these ones:

[Stay off the heap!](https://stackoverflow.com/a/25613483/1676266)

Your code is producing memory leaks.