vshymanskyy/TinyGSM #116: USP socket support

hpeterka, 2017-12-19:
Hello,
is there any reason why this great library doesn´t support UDP sockets? Thanks!

SRGDamia1, 2017-12-19:
I think it's on the TO-DO list.

vshymanskyy, 2018-02-03:
It is on TODO list

closed 2018-02-03 by vshymanskyy

jpmeijers, 2019-04-26:
I was wondering the same thing today when I realised that I will have to do an NTP query to get the time as it's not guaranteed that my Ublox modem gets the time from the network.

UDP in this library would allow us to use:
https://www.arduino.cc/en/Tutorial/UdpNtpClient

SRGDamia1, 2019-04-26:
It's still on the to-do list.

SunboX, 2021-04-21:
Is it still on the To-Do list?

jpmeijers, 2021-04-21:
I can't remember exactly if I ended up implementing this myself or not. My code does: 
```
TinyGsmClient udpClient(modem, 0, UDP);
udpClient.connect(NTP_SERVER, NTP_PORT);
udpClient.write(packetBuffer, NTP_PACKET_SIZE);
while(i < NTP_PACKET_SIZE) {
    int c = udpClient.read();
    ....
}
udpClient.stop();
```

Using my SARA-G450 fork of the library, so it is possible that I implemented the UDP client for this specific modem:
https://github.com/jpmeijers/TinyGSM/blob/g450-relative-incl/src/TinyGsmClientSaraG450.h

SunboX, 2021-04-21:
Thx, I will take a look. Commands might be a bit different for the SIM800 I'm using, but I guess I can figure that out. General library support would be really nice. :)

SRGDamia1, 2021-04-22:
UDP is still on my "someday" list.  But unfortunately, it's still just "someday.."

capedra, 2023-01-22:
@gonzabrusco 
Sure, why not?
[image]