vshymanskyy/TinyGSM #119: Get request over HTTPS fires only once for SIM808

hubaksis, 2018-01-09:
If you use the provided example HttpsClient and try to make two requests without reading headers and the response, the request will be made only once.

In my case I need to make get requests over https. I don't care about response as I need only responseStatusCode. Only the first request works. I am not sure whether it will work if you read response headers and the response body.

Here is the log from the good request:

Request | Response
-- | --
AT+CIPSSL=1 | OK
AT+CIPSTART=1,"TCP","mysite.com",443 | OK
  | 1, CONNECT OK
AT+CIPSEND=1,3          >GET | DATA ACCEPT:1,3
AT+CIPSEND=1,1          > | DATA ACCEPT:1,1
AT+CIPSEND=1,68            >/path?key=12&data=34 | DATA ACCEPT:1,68
AT+CIPSEND=1,9          > HTTP/1.1 | DATA ACCEPT:1,9
AT+CIPSEND=1,2          > | DATA ACCEPT:1,2
AT+CIPSEND=1,6          >Host: | DATA ACCEPT:1,6
AT+CIPSEND=1,26          >"mysite.com" | DATA ACCEPT:1,26
AT+CIPSEND=1,1          >: | DATA ACCEPT:1,1
AT+CIPSEND=1,3          >443 | DATA ACCEPT:1,3
AT+CIPSEND=1,2          > | DATA ACCEPT:1,2
...and so on
... get status, it's 200
AT+CIPSHUT | SHUT OK

When a second response is made, it looks like this:

Request | Response
-- | --
AT+CIPSEND=1,3 | ERROR
AT+CIPSEND=1,1 | ERROR
AT+CIPSEND=1,68 | ERROR
AT+CIPSEND=1,9 | ERROR
AT+CIPSEND=1,2 | ERROR
AT+CIPSEND=1,6 | ERROR
AT+CIPSEND=1,26 | ERROR
AT+CIPSEND=1,1 | ERROR
AT+CIPSEND=1,3 | ERROR
...and so on

So, basically the library tries to send data using CIPSEND without actual data (probably it's OK as http.stop() was called) and without CIPSSL and CIPSTART in the beginning.

CIPSSL and CIPSTART  are not sent because HttpClient doesn't call iClient->connect any more.
Firstly, http.connectionKeepAlive() sets protected variable iConnectionClose = false;
Then, when http.get is made for the second time, it calls http.startRequest. 
startRequest has a condition 
`if (iConnectionClose || !iClient->connected()) { call connect }`
iConnectionClose - false, and iClient->connected() returns true.
Here is this method from TinyGsmClientSIM800
```
virtual uint8_t connected() {
    if (available()) {
      return true;
    }
    return sock_connected;
  }
```
I didn't have a chance to go deeper as I am not sure where the error is: HttpClient or TinyGSM

As for me - I changed the following line in the HttpClient::startRequest:
`if (iConnectionClose || !iClient->connected())`
to 
`if (true)`

Do you know where the actual issue is and how to fix it?
Thanks.

Daniel-dk, 2018-01-18:
it looks like the connected method shouldnt just return true if data is available , it should probably check that teh socket is still connected using "AT+CIPSTATUS " like "bool modemGetConnected(uint8_t mux)" does, assuming one knows which mux was used for this connection ( are there cases where multiple concurrent connections/sockets  are used that justifies using CIPMUX=1 and complicating the system ? )

hubaksis, 2018-02-04:
Before I (or someone else) figure out how to check if the modem connected to the website, I reversed changes made in HttpClient and changed `virtual uint8_t connected()` in `TinyGsmSim800 - > GsmClient` to always return false.

vshymanskyy, 2018-03-13:
There's a concept of "partially connected" socket, which it is when we have still some data to process.
I think the proper solution would be to call `client.stop()`, not only `http.stop();`.
But I also think that this should be done by HttpClient automatically (don't see any reasons not to...).

vshymanskyy, 2018-03-13:
And indeed:
```
void HttpClient::stop()
{
  iClient->stop();
  resetState();
}
```
I now realize that I can fix this by just clearing the input buffer on stop()...

vshymanskyy, 2018-03-13:
@hubaksis please try with the latest master, and ping me back if still have this trouble.

closed 2018-03-13 by vshymanskyy

hubaksis, 2018-03-14:
@vshymanskyy Sorry, can't confirm. Don't have SIM808 any more. If I get it I'll check.

vshymanskyy, 2018-03-14:
Any SIM8xx would do ;)

wbmrcb, 2018-08-25:
@vshymanskyy Sorry, still have this issue on SIM800 (after trying two POST requests)

commit c8d71b3b (2018-03-13): Clear buffer on stop(). Fix #119