vshymanskyy/TinyGSM #120: [Question] Are there a limit on sending a post request using SIM808?

hubaksis, 2018-01-10:
I am sending text files over https manually using following method:
```
http.beginRequest()
..
http.post("url")
http.sendHeader("Content-Type", F("multipart/form-data; boundary=--=123456"));
http.sendHeader("Content-Length", fileLength + empty_request_length);
http.beginBody()
http.println("----=123456");
http.println("Content-Disposition: form-data; name=\"file\"; filename=\"myfile.txt\"");
http.println("Content-Type: text/plain");
http.println();
//Read file and 
//send file line by line 
// http.print(buffer);

http.println();
http.endRequest()
```

Buffer's size is 100 bytes.
I managed to send files with the size of 10,000 bytes. However, some of the http.print(buffer) returned error, that is why I make several tries while sending a line.

I noticed that sometimes SIM808 modem reboots, sometimes it sends data correctly. 

My questions:
Do you know how actually data are sent when I do http.print(buffer)? 
Do they get stored in some buffer in SIM808 or they are sent to the server immediately?
In there is a buffer, then it should be some limit on the file. 
If no buffer - far as I understand - there should be no limit on the client side. The only limit is the buffer's size. Do you know what is it?

Thank you.

hubaksis, 2018-01-10:
Ok, played a bit with my code and found out that the size of buffer that can be sent in one call is 1460. (Fond confirmation here: http://www.python-exemplary.com/download/sim800_series_tcpip_application_note_v1.00.pdf, page 20, "6.3 Query Available data length to be sent").
It varies on the network, so I left 1000 to be safe.

And I managed to send a file of 200Kb. That is why, I assume that data are sent directly and file size is not limited on the client side.

vshymanskyy, 2018-02-03:
Yes, I can confirm.

closed 2018-02-03 by vshymanskyy