vshymanskyy/TinyGSM #121: intermittent intitialisation timeouts

jameshowe, 2018-01-15:
Finding calls to `init` time out even though I can see (via Serial monitor) that the correct response comes back e.g.

**Code**
```
#include <SoftwareSerial.h>
SoftwareSerial SerialAT(PIN_RX, PIN_TX);
...
void setup() {
  // Set console baud rate
  Serial.begin(9600);
  delay(10);
  
  SerialAT.begin(9600);
  delay(3000);

  Serial.println(F("Initializing modem..."));
  if (!modem.init()) {
    Serial.println(F("FAILED!"));
    return;
  }
}

```
**Log**
```
Initializing modem...
AT

OK
AT

OK
AT

OK
... (10s later)
FAILED!
```

It would appear as if the library isn't processing the response correctly (which *sounds* like a baud rate issue) however have tried various baud rates and doesn't seem to fix it.

Interestingly, if I switch to a different library e.g. [Adafruit_FONA](https://github.com/adafruit/Adafruit_FONA) and try commands it works fine so only appears to be a problem with this lib.

hubaksis, 2018-01-18:
Try an example
```
uint32_t modemSpeed = TinyGsmAutoBaud(SerialAT);
Serial.print(F("Modem responded at speed "));
Serial.println(modemSpeed);
//then restart instead of init just in case
if (!modem.restart()) {
    Serial.println(F("FAILED!"));
    return;
  }
```

jameshowe, 2018-01-23:
@hubaksis I was previously using the `TinyGsmAutoBaud` function, this isn't really the root of the problem for me. The library sometimes works fine but over time it just gets to the point where the library becomes unreliable (as per the example). I initially thought perhaps it was a RAM issue, however, after re-writing a sketch and pulling in the bare minimum it was working and then you'd start getting garbage down the Serial monitor for some commands.

Just worryingly unreliable, the Adafruit_FONA library seems to be _far_ more reliable & stable (at least for me).

hubaksis, 2018-01-30:
That's really wired. I have never faces issues on this stage. Well, actually it was only once when I chose a wrong CPU frequency when uploading a sketch and I messed up with my wiring a bit.
If you can see 'OK' in debug log then the library also should see it.

Can you show the full code and full log when it happens? Does it happen when you turn on the power or in some time? What modem are you using?

vshymanskyy, 2018-03-09:
Cannot confirm, works as expected for me and many others...
What exactly is your hardware and wiring? Please reopen if still have this issue.

closed 2018-03-09 by vshymanskyy