vshymanskyy/TinyGSM #122: Can you explain the function waitResponse() ?

quan1506, 2018-01-17:
I read your library and I saw that in TinyGsmClientSim800.h having a waitResponse() function which used many times in implementing other functions. Can you explain me what is that function used for in detail ? 
Im very appreciate your helping. Thank you

hubaksis, 2018-01-18:
Not sure what detailed level you are after, so I sorry if I tell you obvious things.
This function waits for response from the modem.
Communication with modem is done by using AT commands. It means that you are sending AT[+ some command] and it responds to you in some way (you can find it your modem manual).
Lets have a look at the modemGetConnected
```
bool modemGetConnected(uint8_t mux) {
    sendAT(GF("+CIPSTATUS="), mux);
    int res = waitResponse(GF(",\"CONNECTED\""), GF(",\"CLOSED\""), GF(",\"CLOSING\""), GF(",\"INITIAL\""));
    waitResponse();
    return 1 == res;
  }
```
` sendAT(GF("+CIPSTATUS="), mux);` - sends AT+CIPSTATUS=1
The first `waitResponse` waits for the modem response with the specified text. As soon as it gets it (or timeout happens) - the program continues.
The second `waitResponse();` waits for the text `OK` or `ERROR` with 1 second timeout.

vshymanskyy, 2018-01-31:
Thanks @hubaksis

closed 2018-01-31 by vshymanskyy

sajibkarmaker, 2019-07-22:
which file contains the definition of waitResponse(); function

eabase, 2019-07-22:
It was explained to me [here](https://github.com/vshymanskyy/TinyGSM/issues/189#issuecomment-513299112).

aufdenkampe, 2020-08-30:
@SRGDamia1 & @eabase, I appreciate the explanation of the `waitResponse()` function, but I'm still interested in where it is defined (asked by @sajibkarmaker). 

In particular, I'm interested in knowing whether commands like the following can only be applied to modem objects in this library or whether modems inherit those commands from elsewhere:
```cpp
thisModem().sendAT(GF("+IPR="), baud);
thisModem().waitResponse();
```

Do `sendAT()` and `waitResponse()` work on any serial stream object?

I've dug around a bunch in this library and can't find where they are defined. I've also dug a good bit online for other examples of these functions and I'm coming up empty-handed.

hubaksis, 2020-08-30:
@aufdenkampe 
sendAT() is definded here: https://github.com/vshymanskyy/TinyGSM/blob/ad437a5a5166960794a190b0d6ea945282cb992a/src/TinyGsmModem.tpp#L27
waitResponse() depends on the particular modem and defined for each implementation, for example here: https://github.com/vshymanskyy/TinyGSM/blob/ad437a5a5166960794a190b0d6ea945282cb992a/src/TinyGsmClientSIM5360.h#L600 
or here: https://github.com/vshymanskyy/TinyGSM/blob/ad437a5a5166960794a190b0d6ea945282cb992a/src/TinyGsmClientSIM800.h#L619

aufdenkampe, 2020-08-31:
@hubaksis, thanks! I feel embarrassed that I didn't find those myself!
I appreciate that help.

SRGDamia1, 2020-08-31:
The sendAT and waitResponse functions should be usable by any TinyGsmModem object, which in turn can be created using any Stream object (Serial, SoftwareSerial, AltSS, etc).

The functions are called within the library using the "thisModem" object because all of the modems are created using CRTP and it the function needs to be pointed to its own specific class implementation of the generic modem.  You don't need to any of that in your own program because you will have already created a specific object and all the templates will have been resolved already by the compiler.

The waitResponse is a pretty spiffy function. 
- With no arguments it waits for 1s looking for "OK" or "ERROR", returning 1 for the former and 2 for the latter.  This is useful for sending the modem all kinds of commands that you expect it to reply to with an OK.
- Given up to 5 (or 6 for a few) constant character strings, it will wait 1s for any of those instead and return the number of the response it got.  Note that it will stop waiting after the first of any of them provided strings arrive.
- Given an uint32_t, it will wait that number of ms for the same responses.
- Given a String input, in addition to the others, it will append whatever comes after the specified text onto your String.

I didn't think up the original function, I've just implemented it a bunch of times.

aufdenkampe, 2020-08-31:
@SRGDamia1, thanks for this additional explanation. I'm specifically exploring this as I start exploring how to use a MultiTech mDot LoRa modem, which I want to eventually pull into EnviroDIY ModularSensors.

So, can I create a `TinyGsmModem ` object just from my serial1 object, without yet building a full .h file for the mDot? If so, how? I can't find an example of a constructor.

I must admit that I am quite confused by this TinyGSM repo, probably in part because of it's use of Templates.

SRGDamia1, 2020-08-31:
@aufdenkampe Um., well, no.  At least, not in the way I think you mean.  You can't create a TinyGsmModem object by itself.  You have to implement the template and create a specific type of modem.  In all the examples and code you've probably used it's felt like you created a TinyGsmModem object, but you didn't.  In TinyGsmClient.h there are a bunch of typedefs that define a specific modem the generic name as the specific type so it's easier and faster to type.  Which type is used depends on which you've defined (ie, with ```#define TINY_GSM_MODEM_XX```)

If the *only* functions you want to use are sendAT and waitResponse, you could just pick any type that use the right line endings (most are CRNL, XBee is CR only), and create that modem type and then use just those two functions.  Doing that would speed up make it easy to send AT commands to your mDot, but it's a waste and you might as well just implement those two functions within your code and not bother with the library.

If you want to do something useful, like say create a _Client_ you'd need to create a full header and implementation.

aufdenkampe, 2020-08-31:
@SRGDamia1, OK. That's the answer I had expected, and I'm definitely heading toward working on a full implementation of the mDot in TinyGSM. 

At the moment, however, I'm still learning and I had a small hope from your last response that there might be an intermediate step. No worries. This has all been very helpful.