vshymanskyy/TinyGSM #123: Help me clarify the SendUSSD function

quan1506, 2018-01-23:
I want to use SendUSSD function in TinyGSMClinet800.h to check my account's service such as balance and the amount of megabytes remain. At first, I use AT command "AT+CUSD=1,\"*101#\" with *101# is my code from the local service. It returns right thing. But when I use SendUSSD function with that code, it returns nothing. So I came back to my above AT command but now it returns +CUSD: 2 and that 's all, not return the same thing like I did at first. I dont know whether your function change the work of my Sim or not. Hope you can clarify me. Thank you. I know when using some thing I must understand it clearly, but I think just give it a try and look what it does  and then now I dont know to fix it :( So it very nice if you show me the point to turn my Sim 808 back to normal.

hubaksis, 2018-01-30:
Try to turn on debug and check what is the actual response from the modem.

In the examples uncomment this line

```
#define DUMP_AT_COMMANDS 

```

vshymanskyy, 2018-01-31:
@quan1506 sorry to hear that. First of all, follow hubaksis idea.
Secondly, do a Factory Reset (README provides instructions)

quan1506, 2018-02-02:
Thank you all, I solved this problem.

hubaksis, 2018-02-02:
@quan1506 Can you please share what was the cause of the problem and how you resolved it? Thanks.

closed 2018-02-03 by vshymanskyy

spa-sam, 2018-03-09:
I have the same problem. Dumped to factory settings - did not help. How did you solve the problem?

vshymanskyy, 2018-03-09:
@quan1506 ?

quan1506, 2018-03-10:
Honestly, I don't do anything, in the next day, I use the AT command for checking the balance and it works again. I have no idea why :))