vshymanskyy/TinyGSM #124: How to restart modem immediately after fail anything ?

matriawan, 2018-01-28:
looking this function originally in block of "void setup()"

if (!modem.waitForNetwork()) {
Serial.println(" fail");
modem.restart(); // added
while (true);
}

but i face the challenge when the fail message in the block of "loop()" like error message "Connecting to test.mosquitto.org fail", how to make the modem restart like the beginning ?

matriawan, 2018-01-29:
waitForNetwork function always skipping, my solution modify "TinyGSMClient.h" library:

` bool waitForNetwork(unsigned long timeout = 60000L) {
    for (unsigned long start = millis(); millis() - start < timeout; ) {
      RegStatus s = getRegistrationStatus();
      if (s == REG_OK_HOME || s == REG_OK_ROAMING) {
        return true;
      }else {
		return false;
	  }
      delay(60000L);
    }
  }`

hubaksis, 2018-01-30:
Not sure what you are trying to achieve.
`waitForNetwork` actually waits for the network (modem is registered in the network and nothing else)
`gprsConnect` connects to GPRS. Firstly it closes connection (if any) and starts a new one.
If an error occured with the 'connecting failed' message, there are several possibilities: network failure, gprs connection failure or server failure. If the modem is connected to the network - waitForNetwork will be always skipping.

hubaksis, 2018-01-30:
By the way, the original method will wait for a connection up to 60 seconds. If it's connected in 5s - it will exit in 5 seconds with the success. Your code will wait for exactly 60s. 
I also recommend you to define DUMP_AT_COMMANDS (see HttpClient example) to understand what is actually going on.

closed 2018-01-30 by matriawan

capedra, 2022-04-05:
Alright, I decided to eliminate the timeout just to simplify things up and as an alternative:
```
bool waitForNetwork() {
  RegStatus s = modem.getRegistrationStatus();
  while (s == REG_SEARCHING) {
    s = modem.getRegistrationStatus();
  }
  if (s == REG_OK_HOME || s == REG_OK_ROAMING)
    return true;
  return false;
}
```

drChelo, 2022-08-19:
Hi @capedra ! Is this function useful to check that both the modem and the serial comm are working correctly? I want to make a failsafe inside the loop() to check every 5 or 10 min that both things are working fine.