vshymanskyy/TinyGSM #125: Reduce time of connecting GPRS

quan1506, 2018-02-02:
I have a question on how to reduce the connecting time to GPRS using the function _gprsConnect(apn, user, pass)_ . As I read in datasheet of Sim808, the way to connect GPRS is simply like in the picture. But in your function I see many other AT commands. So, is there anyway to reduce time of connection by reducing some AT commands that does nothing with the SIM 808. Because when I use that function, I see that it take 3 times to run properly and about ~20s for a successful connecting,

[image]

vshymanskyy, 2018-02-03:
Thanks, I'll check.. But from my experience, the recommended method of setting up the connection for SIM8xx would be exactly as implemented in the library.

vshymanskyy, 2018-03-14:
Ok, after some testing, I can say that with less commands you can hardly get the lower cold startup time.
It might be a small improvement, but at cost of stability - so I'd leave it as is.
Most of the commands are really fast in that sequence.

closed 2018-03-14 by vshymanskyy