vshymanskyy/TinyGSM #127: stuck in 1, connect ok

Akhail, 2018-02-04:
Hi i use arduino + LiquidCrystal_i2c + tinyGSM + pubsubclient stuck alway i use liquidcrystal remove this and it work well

hubaksis, 2018-02-04:
Hi @Akhail 
Sorry, can't understand a thing from your message.
Please explain in details what is the issue and attach some code and logs with the issue.

Akhail, 2018-02-05:
the log is
```
Initializing modem...
AT

OK
AT+CFUN=0

+SAPBR 1: DEACT

+PDP: DEACT

+CPIN: NOT READY

OK
AT+CFUN=1,1

OK
AT
⸮⸮AT
AT

OK
AT&FZ
AT&FZ

OK
ATE0
ATE0

OK
AT+CPIN?

ERROR
AT+CPIN?

RDY

+CFUN: 1

ERROR
AT+CPIN?

+CPIN: READY

+CPIN: READY

OK
Waiting for network..
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

Call Ready

+CREG: 0,2

OK
AT+CREG?

SMS Ready

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,1

OK
 NETWORK OK
Connecting to internet.movistar.ve
AT+CIPSHUT

SHUT OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","internet.movistar.ve"

OK
AT+SAPBR=3,1,"USER",""

OK
AT+SAPBR=3,1,"PWD",""

OK
AT+CGDCONT=1,"IP","internet.movistar.ve"

OK
AT+CGACT=1,1

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"10.212.239.33"

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

OK
AT+CSTT="internet.movistar.ve","",""

OK
AT+CIICR

OK
AT+CIFSR;E0

10.212.239.33

OK
AT+CDNSCFG="8.8.8.8","8.8.4.4"

OK
 CONNECTION OK
END SETUP
RETRY CONNECT
Connecting to broker 159.89.142.180
Señal: AT+CSQ

+CSQ: 30,0

OK
30
Operador: AT+COPS?

+COPS: 0,0,"73404"

OK
73404
AT+CIPSSL=0

OK
AT+CIPSTART=1,"TCP","159.89.142.180",1883

OK

1, CONNECT OK
```

the code is

network.h
```cpp
#ifndef NETWORK_H_
#define NETWORK_H_

// #define PRODUCTION
#define TINY_GSM_MODEM_SIM800
#define TINY_GSM_RX_BUFFER 512

// Includes
#include <Arduino.h>
#include <inttypes.h>
#include <SoftwareSerial.h>
#include <PubSubClient.h>
#include <TinyGsmClient.h> 

// General config
#define BAUDRATE 57600

// Operator config
#define OPERATOR 2 // Digitel, movistar, movilnet
#define APN_USER ""
#define APN_PASSWORD ""

// Authentication
#define CLIENT_ID "MH01"
#define CLIENT_USER ""
#define CLIENT_PASSWORD ""
#define COMPANY "tux"

// BROKER data
IPAddress BROKER_IP(159, 89, 142, 180);
#define BROKER_PORT 1883

// Ports and configs
#define RX_PORT 13
#define TX_PORT 12

// Values
#if OPERATOR == 1
#define APN_NAME "internet.digitel.ve"
#elif OPERATOR == 2
#define APN_NAME "internet.movistar.ve"
#else
#define APN_NAME "int.movilnet.com.ve"
#endif

SoftwareSerial SerialSIM(RX_PORT, TX_PORT);

#ifndef PRODUCTION
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialSIM, Serial);
  TinyGsm modem(debugger);
#else
  TinyGsm modem(SerialSIM);
#endif

TinyGsmClient client(modem);
PubSubClient mqtt(client);

void initNetwork(void);
void mqttConnect(void);
void mqttCallback(char*, byte*, uint16_t);

long lastReconnectAttempt = 0;

void mqttCallback(char *topic, byte *payload, uint16_t len) {
    #ifndef PRODUCTION
    Serial.print("Topic ");
    Serial.println(topic);
    Serial.print("Payload ");
    Serial.write(payload, len);
    Serial.println();
    #endif
    payload[len] = '\0';
    String msg = String((char*)payload);
}

void initNetwork() {
    Serial.begin(BAUDRATE);
    delay(10);
    SerialSIM.begin(BAUDRATE);

    #ifndef PRODUCTION
    Serial.println("Initializing modem...");
    #endif

    if(!modem.restart()) {
        #ifndef PRODUCTION
        Serial.println(" INITIALIZE FAIL");
        #endif
        while(true);
    };
    delay(10);

    #ifndef PRODUCTION
    Serial.println("Waiting for network..");
    #endif

    if(!modem.waitForNetwork()) {
        #ifndef PRODUCTION
        Serial.println(" NETWORK FAIL");
        #endif

        while(true);
    }

    #ifndef PRODUCTION
    Serial.println(" NETWORK OK");
    Serial.print("Connecting to ");
    Serial.println(APN_NAME);
    #endif

    if(!modem.gprsConnect(APN_NAME, APN_USER, APN_PASSWORD)){
        #ifndef PRODUCTION
        Serial.println(" GPRS FAIL");
        #endif

        while(true);
    }
    #ifndef PRODUCTION
    Serial.println(" CONNECTION OK");
    #endif

    mqtt.setServer(BROKER_IP, BROKER_PORT);
    mqtt.setCallback(mqttCallback);
    #ifndef PRODUCTION
    Serial.println("END SETUP");
    #endif
}

void mqttConnect() {
    if (mqtt.connected()) {
        mqtt.loop();
        return;
    }

    unsigned long time = millis();
    if (time - lastReconnectAttempt < 10000L) return;

    #ifndef PRODUCTION
    Serial.println("RETRY CONNECT");
    #endif

    lastReconnectAttempt = time;
    

    #ifndef PRODUCTION
    Serial.print("Connecting to broker ");
    Serial.println(BROKER_IP);
    Serial.print("Señal: ");
    Serial.println(modem.getSignalQuality());
    Serial.print("Operador: ");
    Serial.println(modem.getOperator());
    #endif
    
    if(!mqtt.connect(CLIENT_ID)) {
        #ifndef PRODUCTION
        Serial.println(" MQTT CONNECT FAIL");
        Serial.print("State: ");
        Serial.println(mqtt.state());
        #endif
        return;
    }

    #ifndef PRODUCTION
    Serial.println(" MQTT OK");
    Serial.println("subscribe to test topic");
    #endif

    mqtt.publish(COMPANY "/machines/" CLIENT_ID, "System ready");
    mqtt.subscribe(COMPANY "/machines/" CLIENT_ID "/#");
    
    if(mqtt.connected()) lastReconnectAttempt = 0;
}
#endif
```
hardware.h
```cpp
#ifndef HARDWARE_H_
#define HARDWARE_H_

#include <Arduino.h>
#include <inttypes.h>
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27, 16, 2);

void initHardware();
void readKeyboard();
void updateDisplay();

void initHardware() {
    lcd.init();
    lcd.backlight();
    lcd.print("Inicializando...");
}
void readKeyboard() {}
void updateDisplay() {}
#endif
```

vshymanskyy, 2018-03-13:
@Akhail it is not clear what modem model are you using. Is it sim800 or sim900?

vshymanskyy, 2018-03-24:
@Akhail please answer or I'm closing the issue.

marabesi, 2018-04-06:
I am having the same problem running [this](https://github.com/vshymanskyy/TinyGSM/blob/master/examples/MqttClient/MqttClient.ino) example.

The SIM8000 connects fine, the network connects fine but for some reason the sketch doesn't connect to the broker.

[image]

vshymanskyy, 2018-04-06:
I'm closing this as couldn't get more info about the issue.

closed 2018-04-06 by vshymanskyy

jeancaffou, 2019-04-14:
@vshymanskyy Same issue, more information in #265