vshymanskyy/TinyGSM #130: HTTP Post required

abhaysbharadwaj, 2018-02-16:
Hello,
First off, thank you for this wonderful library! I was able to implement MQTT over GSM using this!

I found the code for HTTP GET, but couldn't manage to do an HTTP POST. Are there any functions i can use to achieve this? I found the http.get() but is there a http.post()?

SRGDamia1, 2018-02-16:
You don't really need an http.post() or get function, just create a TCP connection with client.connect(host, port) and then send the raw post request as:

client.print(String(F("POST /your/site/here/ HTTP/1.1")));
client.print(String(F("\r\nHost: your.host.here")));
client.print(String(F("\r\nHeader: Header value")));
client.print(String(F("\r\n\r\n")));
client.print(String(your message body here));

hubaksis, 2018-02-16:
SRGDamia1 is right. 
Personally I had issues using http.post, so had to use the same way

```
... 
  HttpClient http(client, server, port);
...

  http.connectionKeepAlive();
  http.beginRequest();
  int codePost = http.post("/api/blablabla");

  DEBUG_MSG_LINE(F("Returned status after post = "));
  DEBUG_MSG(codePost); 

  http.sendHeader(F("Content-Type"), F("<what you are sending>"));
  http.sendHeader(F("Content-Length"), fileLength);
  http.beginBody();
  http.println(<your data>);   

  http.endRequest();

  int status = http.responseStatusCode();

```

closed 2018-02-17 by vshymanskyy