vshymanskyy/TinyGSM #131: how to connect to ibm bluemix

muzmmilkhan, 2018-02-21:
Good Morning sir 
i have a adafruit fona sim808 and i want to connect to ibm bluemix .please guide me how i can connect to ibm bluemix by using mqtt of  tinygsm .also how to define the 
#define ORG "<yourOrganization>"
#define DEVICE_TYPE ""
#define DEVICE_ID "<yourDeviceID>"
#define TOKEN "<yourToken>"
and use it for ibm bluemix
please help

muzmmilkhan, 2018-02-23:
or  other cloud please replay

SRGDamia1, 2018-02-23:
I don't know anything about IBM Bluemix. This library creates an internet TCP client, so you should be able to use it in the same way you would use a client created by any other library.  Have you looked at the mqtt example?

vshymanskyy, 2018-03-09:
Closing as not related to TinyGSM

closed 2018-03-09 by vshymanskyy