vshymanskyy/TinyGSM #132: can any body explane how to decoding dtmf using stream from modem tinygsm800.h

vkskif, 2018-02-23:
Any body please help, how to using basic functional in tinygsm get char from dtmf decoding sim800l???

hubaksis, 2018-02-24:
I think TinyGSM supports only DTMF send right now.
According the AT command Manual, there is a command that you need: AT+DDET.
```
1. Test Command (check it with your module first): 
AT+DDET=?

Response (this you should get, if not - then it's not supported by your module)
+DDET: (0,1),(0-10000),(0,1),(0,1)
OK

2. Read Command (read current settings)
AT+DDET?

Response
+DDET: <mode>,<interval>,<reportMode>,<ssdet>
OK

3. Write Command (set settings)
AT+DDET=<mode>[,<interval>][,<reportMode>][,<ssdet>]

Response
OK

Unsolicited Result Code (this you will receive when the DTMF signal comes)
1)If <reoportMode> is set to 0
+DTMF: <key> 

2)If <reportMode> is set to 1
+DTMF: <key>,<last time>

Parameters
<mode> Disable or enable DTMF detection control
0 Disable
1 Enable

<interval> The min interval between two same key URC. The range is
0-10000, the default value is 0. unit is ms.

<reportMode> URC report mode
0 Key value reported only
1 Key value and last time are reported, the last time is in ms

<key> Keytone detected, 0-9,*,#,A,B,C,D.
If <ssdet> is 1,Single frequency sound 1400 and 2300 is supported too, when single frequency 1400HZ sound or 2300HZ sound is detected, +DTMF:1400 or +DTMF:2300 is reported

<last time> Duration of keytone playing. unit is ms. 

<ssdet> Single frequency sound detect function on off
 0 Switch off
1 Switch on 

Parameter Saving Mode: AT&W_SAVE

```

So, try send the command `AT+DDET=1,1000,0,0(or 1?)` and then catch when the `+DTMF: <key>` comes to you .

vkskif, 2018-02-24:
Thanks
Idea, get DTMF keys until simbol '*'
I try to use code (add to TinyGsmClientSIM800.h) - I wait RING, set DTMF to 1, answer on RING,  but not work and I don't see any keys in debug mode
is my code 

String getDTMFString()
{
   String resp = "";
   char   c;
   if (waitResponse(10000L, GF(GSM_NL "RING")))
   {
      sendAT(GF("+DDET=1"));
      waitResponse();
      sendAT("A");
      waitResponse(); 
      while (stream.available()) 
      {
         c = stream.read();
         if (c == '*') { break; }
         else 
         {
            resp = resp + c;
         }
      }
      //resp =  stream.readStringUntil('*');
   }
   return resp;
}

Execute as
--------------
if (modem.getDTMFString() != "")
{
    DBG("Hold income RING:", "OK");
}
--------------

Debug out put
--------
RING

+CLIP: "+380911120440",145,"",0,"",0
### Unhandled: RING

+CLIP: "+380911120440",145,"",0,"",0

RINGAT+DDET=1
ATA

OK
TEST SerialMon Output
AT+CMGL="REC UNREAD"

OK
TEST SerialMon Output
AT+CMGL="REC UNREAD"

OK
TEST SerialMon Output
AT+CMGL="REC UNREAD"

OK

NO CARRIER
### Unhandled: NO CARRIER
TEST SerialMon Output
AT+CMGL="REC UNREAD"

OK

May be any body have any good idea - I can try!!!!

hubaksis, 2018-02-24:
First thing that I noticed - you are not receiving OK for `AT+DDET=1` and send ATA instead.
I think there is something in the receive buffer, so wait for "OK".

hubaksis, 2018-02-24:
There is a sketch AT_Debug in tools. Try it first and write all commands manually. Check how it works first.

vkskif, 2018-02-24:
AT command tool OUTPUT
-------------------------
OK
AT
OK

+CPIN: READY

Call Ready

SMS Ready
AT+DDET=?
+DDET: (0,1),(0-10000),(0,1),(0,1)

OK
AT+DDET=1,1000,0,0
OK

RING

+CLIP: "+380911120440",145,"",0,"",0

RING

+CLIP: "+380911120440",145,"",0,"",0

RING

+CLIP: "+380911120440",145,"",0,"",0
ATA
OK

+DTMF: 1

+DTMF: 2

+DTMF: 3

+DTMF: 4

+DTMF: 5

+DTMF: 9

+DTMF: *

+DTMF: #

+DTMF: 0

NO CARRIER
-----------

From my point it's OK
How to implement as function !?

vkskif, 2018-02-24:
I write new version code - but not work too

String getDTMFString()
{
   String resp = "";
   //String& data='';
   String data;
   //const String& D1;
   char D1[11]  = "+DTMF: 1"  ;
   char D12[11] = "NO CARRIER";
   char   c;
   if (waitResponse(10000L, GF(GSM_NL "RING")))
   {
      //sendAT();
      waitResponse(3000L);
      sendAT(GF("+DDET=1,1000,0,0; ATA"));
      delay(1000);
      waitResponse(3000L);
      sendAT("A");
      waitResponse(3000L);
      int curr_ms = millis();
      while ((millis()<(curr_ms+80000)) && ((stream.available() > 0)) )
      {
        int a = stream.read();
    	if (a <= 0) continue;
	data += (char)a ;
	if (data.endsWith(D1)) { resp += '1'; DBG("### HOLD DTMF:", data); }
	/* 
        if (data.endsWith('+DTMF: 2')) { resp += '2'; }
	if (data.endsWith('+DTMF: 3')) { resp += '3'; }
	if (data.endsWith('+DTMF: 4')) { resp += '4'; }
	if (data.endsWith('+DTMF: 5')) { resp += '5'; }
	if (data.endsWith('+DTMF: 6')) { resp += '6'; }
	if (data.endsWith('+DTMF: 7')) { resp += '7'; }
	if (data.endsWith('+DTMF: 8')) { resp += '8'; }
	if (data.endsWith('+DTMF: 9')) { resp += '9'; }
	if (data.endsWith('+DTMF: 0')) { resp += '0'; }
	if (data.endsWith('+DTMF: #')) { resp += '#'; }
	if (data.endsWith('+DTMF: *')) { break; } */
        if (data.endsWith(D12)) { break; DBG("### HOLD DTMF:", data); } 
      }
      //resp =  stream.readStringUntil('*');
   }
   return resp;
}

Any body help, whats wrong?

hubaksis, 2018-02-25:
1. You should never compare millis with any value. It's unsigned long and the comparison should be done only by time difference, i.e. `millis() - old_time < allowed_timeout`
2. `stream.available()` will be more than 0 if you have any unprocesses data in the stream. So, your condition won't work.
3. To insert code read 'Syntax highlighting' from [this page](https://guides.github.com/features/mastering-markdown/) 

Try something like this (I didn't check it on a real modem, but you'll get the idea)

```
String getDTMFString()
{
	if (waitResponse(10000L, GF(GSM_NL "RING")))
	{
		sendAT(GF("+DDET=1,1000,0,0;"));
		waitResponse();
		sendAT(GF("A"));
		waitResponse();

		int curr_ms = millis();
		bool shouldStop = false;
		bool timeOut = false;
		//allow max 80s for communitation
		while (millis() - curr_ms < 80000L && !starReceived && !timeOut)
		{
			//wait max 30 s for a key press
			if (waitResponse(30000L, GF(GSM_NL "+DTMF:")) != 1) {
			  DEBUG_MSG(FPSTR("Timeout waiting for a key"));
			  timeOut = true;
			}
			else
			{
				char result = ' ';
				result = stream.readStringUntil('\n').toInt();
				//Print to debug result
				//add to your result string
				//Check if it's '*' then set starReceived = true
				waitResponse();
				
			}
			
		}		
	}
	return resp;
}
```

vkskif, 2018-02-26:
Thanks, I rewrite code and get work example
--------------
String getDTMFString()
{
	String resp, out = "";
        boolean starReceived = false;
        if (waitResponse(10000L, GF(GSM_NL "RING")))
	{
		sendAT(GF("+DDET=1,1000,0,0;"));
		waitResponse();
		sendAT(GF("A"));
		waitResponse();

		int curr_ms = millis();
		bool shouldStop = false;
		bool timeOut = false;
		//allow max 80s for communitation
		while (millis() - curr_ms < 80000L && !starReceived  && !timeOut)
		{
			//wait max 30s for a key press
			if (waitResponse(30000L, GF(GSM_NL "+DTMF:")) != 1) {
			  DBG(FPSTR("Timeout waiting for a key"),"");
			  timeOut = true;
			}
			else
			{
				resp = stream.readStringUntil('\n'); //.toInt();
                                resp.trim();
                                out += resp;
                                DBG(FPSTR("Hold Output DTMF:"), resp);
                                DBG(FPSTR("Hold Output:"), out);
                                if (resp == "*") {  starReceived = true; }
				//waitResponse();
			}
			
		}		
	}
	return out;
}
--------------------------
Thanks, any body can use this code for catch DTMF!!!

vshymanskyy, 2018-03-09:
@vkskif thanks I'll try to integrate it!

closed 2018-03-09 by vshymanskyy

LeeoMariani, 2020-10-31:
Hello, was DTMF receive integrated? I did not found on the lib. thx

SRGDamia1, 2020-11-02:
@LeeoMariani - No.

Mamdouh-Freelancer, 2021-04-11:
Hello all,
  Can this module handle any methods of DTMF? Currently my mobile operator network doesn't pass DTMF as key tones, It sent as digital signals, in another word If a caller (A) pressed any key receive (B) doesn't hear any tones during the call.

SRGDamia1, 2021-04-28:
This library doesn't currently support *receiving* DTMF at all, only sending it.