vshymanskyy/TinyGSM #133: Library not working with ESP32 hardware Serial2

kennythum, 2018-02-24:
Hello there. Firstly, i would say great work and thank you for the wonderful work! I encountered a problem that the library is not working with the ESP32 on hardware Serial2. Somehow I try sending AT command via Serial2 and it works! But running with tinyGsm library it gives unknown characters feedback, or no response sometimes. I did try to use Serial1 (UART1), but Serial1 give nothing at all, no responding. Hence, only Serial2 were able to communicate with the GSM module...just when using with tinyGsm library, it can't work. Anyone could help? Really appreciate.

hubaksis, 2018-02-24:
@kennythum 
What GSM module do you use?

kennythum, 2018-02-24:
@hubaksis Sorry for not providing any details:
ESP32 board : ESP-WROOM-32 (revision :1)
GSM Module : SIM800C
IDE : Arduino 1.8.4

hubaksis, 2018-02-24:
Can you please show your connection between the module and ESP32?

kennythum, 2018-02-24:
I don't think its the hardware or the connection problem. Because I can communicate it with other devices...
But here you go:
[https://imgur.com/P7Gr3X8](url)

hubaksis, 2018-02-24:
@kennythum 
Actually, it can be, I am just not sure yet :)

Thank you for the connections. Now I can see that it's the same as mine.

I have the same issue when I used communication through TX and RX pins on ESP8266.
ESP8266 and ESP32 have 3.3V logic level.
SIM modules (probably, just my guess) are using a different logic level.

So, I checked the communication and found out that the modem can read and understand what the controller is sending, it responds back, but the controller can't understand the response.

So, there are two options you can check:
1. Use different software pins and use SoftwareSerial library. It worked for me without any issues.
2. Use TTL logic level shifter. I am going to do it later to check why the communication is not working on TX RX. Voltage divider didn't work for me.

hubaksis, 2018-02-27:
I did a test and connected a modem to ESP8266's Serial using TTL logic level converter.
It works.
Without level shifter - no.
By the way, ESP doen't start if TX pin is connected. Probably it should be pulled up high.

hubaksis, 2018-02-27:
@kennythum 
GSM module can understand 3.3-5V logic, but ESP can't. I did this test on ESP8266 and modules SIM808 and 5320.
TTL logic converter looks like [this](http://www.icstation.com/logic-level-converter-level-converter-p-4308.html)
My connections are:
Modem RX - RXI
Modem TX - TXO

ESP TX - RXO
ESP RX - TXI

I had to disconnect ESP TX for the module to start and then connect it back.

If tinyGSM return gibberish - it means that the library chose a wrong connection speed. What is the result of the `TinyGsmAutoBaud`?

If you can send commands and see the answer in the console window, then the library also will be able to do so. It's really simple and this part of the code can't work.

What the library does when you call `int modemSpeed = TinyGsmAutoBaud(SerialAT);`
1. It selects the next speed from the array `{ 115200, 57600, 38400, 19200, 9600, 74400, 74880, 230400, 460800, 2400, 4800, 14400, 28800 };`
2. It sets the port speed by calling `SerialAT.begin(rate);`
3. It sends command "AT" to the modem and waits for an answer. If the speed is modem working speed, then the modem will understand the command and will respond to it.
4. If the answer is "OK" then the chosen speed is correct.

It's not a rocket science and I am 90% sure that this code can't fail. It's really simple and there are not vital dependencies.

So,
0. Add a delay to the setup to wait until the module starts or power it up separately (on different modules I noticed a delay until it start to respond).
1. Can you please provide the code that you are using to communicate with the modem
2. The output from this code.
3. Try to load AT_Debug sketch from tools to the ESP and check whether you can communicate with the modem.

bionicbone, 2018-03-03:
I had an issue with it and ended up replacing the Return 0 at the end with Return 19200. I  didn't dig in, I'm using a nano and could have been memory issues or something.

ankgt, 2018-03-09:
@hubaksis, @kennythum  

I have a similar configuration like yours where I have connected an ESP32 Doit Devkit V1 board to a SIM800L modem using HardwareSerial (Uart2 connected to pins 16,17).

Issuing AT commands manually using the AT_Debug works fine, although I had to manually set the baud to 9600. The auto detect selected 115200, which did not work.

Using the AT commands I can successfully make a GET request using the correct sequence of commands.

However, running the TinyGSM Diagnostics example results in garbled text results. 

I have seen other issues like this, but not seen any solutions. I am thinking that it should not be a TTL logic level issue, because in that case it should not work with issuing manual AT commands either, am I right?

Any suggestions on what I can try next will be appreciated. Thanks!

vshymanskyy, 2018-03-09:
This is covered in README: https://github.com/vshymanskyy/TinyGSM#esp32-hardwareserial

ankgt, 2018-03-09:
@vshymanskyy Thanks. I should have mentioned that I have already read this and included the extra parameters in Serial.begin() as suggested. It does not help. 

I have not been able to try shifting to Uart1 though, because my USB serial port is connected to it. 

If it is a hardware or Arduino HardwareSerial problem, then should it not work if I am manually sending the AT commands using AT_Debug?

Thanks.

vshymanskyy, 2018-03-09:
@ankgt  when you enter manually, the speed of typing is so slow ;)

vshymanskyy, 2018-03-09:
@ankgt I would suggest posting this bug to ESP32, most probably the issue is there: https://github.com/espressif/arduino-esp32

ankgt, 2018-03-10:
I will do that. 

I still am wondering about the manual commands though. While I agree I cannot type so fast :) if I manually issue the AT commands to make an HTTP GET request, the modem retrieves the data from the site and sends it back to the ESP32 at full speed, and that also works without any errors. Also when I issue the commands through the Arduino serial monitor, it does not send them character by character, but instead sends the entire string once the enter key is pressed. 

So could the issue be the timing between the individual commands being sent then?

ankgt, 2018-03-10:
Just a quick followup. I tried using another library (https://github.com/carrascoacd/ArduinoSIM800L) and modified it slightly as it was designed for use with SoftwareSerial. It works correctly with my ESP32 and HardwareSerial UART2. I can successfully make HTTP GET and POST requests. 

So I don't think the problem is linked to the ESP32 HardwareSerial implementation.

vshymanskyy, 2018-03-10:
@ankgt You're making wrong conclusions ;) It means that the library you mentioned does not cause the same the problem. Buy TinyGSM does not use anything but correct, platform-agnostic Arduino API,
And it is not causing troubles in many other platforms.

mdc4u, 2018-03-11:
I have esp32 and sim800 working perfect. if somebody need some help...

bionicbone, 2018-03-11:
I may do soon, although I'll be setting it up to do FTP in a couple of weeks. I've got it working with a nano so hopefully it won't be so difficult.

kennythum, 2018-03-12:
The funny thing is, I have couples of ESP32 board, 1 is the older version (which i believe, CE1313 marking engraved on it), and 3 was the later version...they all are the revision 1. The older version ESP32 could make connection with the ESP32 hardware serial1 port (with pin 16,17  <--I know pin 16,17 is for serial2, but serial2 could not work, so i reroute serial1 to pin 16,17). Meanwhile, the 3 later version ESP32 chip could not use with the tinyGSM library at all, but they all could make a good AT response by sending commands to the SIM800C! It even can establish connection, HTTP GET/POST without a problem.

ankgt, 2018-03-12:
@vshymanskyy It's possible! I am fairly new to embedded development and still trying to figure things out!

@kennythum I guess my board is a newer one, since I can to establish proper connections using other libraries. I would really like to use TinyGSM though, since its very well tested and has wide functionality.

If you connected using Serial1 port using 16,17, I am assuming you could no longer monitor the running of the program using the terminal from your PC? Or did you try connecting Serial2 to the built in USB-UART converter pins?

ankgt, 2018-03-12:
@mdc4u Yes would love some help! Can you confirm if you are using HardwareSerial with UART2? Is there anything specific you did to get it working?

mdc4u, 2018-03-12:
Well, in my case, I have my project distributed as follows:
UART0 -> usb to debug
UART1 -> (gpio 4 and 2) SIM800 (115200 baud)
UART2 -> (gpio 16 and 17) GPS

in HardwareSerial.cpp

`if (_uart_nr == 0 && rxPin <0 && txPin <0) {
         rxPin = 3;
         txPin = 1;
     }
     if (_uart_nr == 1 && rxPin <0 && txPin <0) {
         rxPin = 4;
         txPin = 2;
     }
     if (_uart_nr == 2 && rxPin <0 && txPin <0) {
         rxPin = 16;
         txPin = 17;
     }
`
everything works correctly.
I have even managed to operate OTA through SIM800 instead of using wifi.

My big problem is that the client, when I move (for example in a car) is randomly disconnected.

Using the same code in a poor nano arduino, the disconnections do not occur. I hope to solve this mystery soon,

I hope I can be useful
regards
and thanks!

Paul

vshymanskyy, 2018-03-12:
@mdc4u 
> managed to operate OTA through SIM800 instead of using wifi

Could you share some code for that? I would like to check it, and maybe optimize the scenario

mdc4u, 2018-03-12:
Well, it's complex, but not at all impossible, and no problem in illustrating with a bit of code,
At this moment the OTA through SIM800 I have implemented with a private project, give me some time to extract that, and make it more universal.

vshymanskyy, 2018-03-12:
@mdc4u would appreciate that!

mdc4u, 2018-03-12:
@vshymanskyy 

sorry for the untidiness I have quickly assembled this example to download .bin server saves it in spiffs and then apply ota.

this works without any problem.

where if I have a problem is to keep a tcp client connected,
(in another part of my project of course nothing to do with OTA)
in my .ino, check connection periodically as any typical tnygsm script, in case the socket has been closed, I try to connect again, if in several attempts does not happen.
well, restart both esp and sim800,
this, applied to an arduino nano or an arduino one, works well for me
but, B U T
this same using an esp32, becomes unstable when I move in a vehicle, is disconnected every time. This does not happen with a nano,
what exactly brought me to esp32 is to be able to use their nucleuses separately, and the blessed OTA

What is promised is debt, here goes

```c++
#include <Update.h>
#define TINY_GSM_MODEM_SIM800
// Increase RX buffer
#define TINY_GSM_RX_BUFFER 1030

const char apn[] = "";
const char user[] = "";
const char pass[] = "";

HardwareSerial SerialAT(1);

#include <TinyGsmClient.h>
#include <CRC32.h>

#include "FS.h"
#include "SPIFFS.h"

#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, Serial);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif
TinyGsmClient client(modem);

const char server[] = "here the server like sample.com.ar";
const int port = 80;

const char resource[] = "/updates/v1_0.bin"; //here de bin file

uint32_t knownCRC32 = 0x6f50d767;
uint32_t knownFileSize = 1024; // In case server does not send it

void setup()
{
    // Set console baud rate
    Serial.begin(115200);
    delay(10);

    if (!SPIFFS.begin(true))
    {
        Serial.println("SPIFFS Mount Failed");
        return;
    }
    SPIFFS.format();
    listDir(SPIFFS, "/", 0);

    // Set GSM module baud rate
    SerialAT.begin(57600);
    delay(3000);

[... 331 lines cut ...]
        }
        if (Update.end())
        {
            Serial.println("OTA realizado!");
            if (Update.isFinished())
            {
                Serial.println("Ota exitoso, reiniciando!");
                ESP.restart();
            }
            else
            {
                Serial.println("Ota no terminó? Algo salió mal!");
            }
        }
        else
        {
            Serial.println("Ocurrió Error #: " + String(Update.getError()));
        }
    }
    else
    {
        Serial.println("Sin espacio suficiente para hacer OTA");
    }
}
```

vshymanskyy, 2018-03-13:
@mdc4u From the first glance, it looks like we can skip the FS part at all, and feed The GSM Client directly to the updater. But! I'll have to check several things ;)
Thanks!

mdc4u, 2018-03-13:
@vshymanskyy 
Hello
Yes, you are absolutely right, feed the updater, it would be great.
Due to my little knowledge about the free updater, my direct feed attempts did not work. And what you saw is the only way I found to make it work. Totally inefficient by the way.

Well ... going back to the topic that brought us.
At last I was able to make the client not disconnect after several hours of operation (sending GPS data) using sim800 ...

in my case the solution was simple.
short cables and well welded. It's more ... now to make sure I've soldier what could be said ... pin with pin

now if ... no more problems ...
my suspicion started when I inspected the talk between the esp and the sim800 ...

the famous "?" they appeared from time to time, and that's when the disaster happened ...

then it appeared,
my amateur radio spirit ... that has to be noise.

I quickly removed the cables and welded pin with pin

ready. in my case.
the problem was solved

I hope it serves someone.

Thank you!!!!

Pablo

vshymanskyy, 2018-03-14:
@kennythum there are multiple confiramtions that this is working.
Closing the issue for now. Please reopen if still have issues, and report it to ESP32 core as well

closed 2018-03-14 by vshymanskyy

ankgt, 2018-03-15:
@mdc4u  Thank you very much for the tip. I have managed to get the library working on Serial1!

kamrulroky, 2018-05-12:
@enerbits  OTA through GSM looks awesome, I am trying to do same thing on esp8266 (4mb flash chip version) with sim800l .the code is for esp32,and i am not sure for "update.h" what is the corresponding library on esp8266. will it work if i change file system objects for esp8266?

vshymanskyy, 2018-05-12:
@kamrulroky probably won't work for ESP8266 this way. I'm preparing an article on how to make this working in an optimal way for ESP8266 and ESP32

vignesh13404387, 2018-11-12:
> @vshymanskyy
>
> sorry for the untidiness I have quickly assembled this example to download .bin server saves it in spiffs and then apply ota.
>
> this works without any problem.
>
> where if I have a [...]

hello, Which controller is suitable for this program.  I am using ESP8266

vignesh13404387, 2018-11-27:
> @enerbits <https://github.com/enerbits> thanx for code sharing.
> I have a code for those who wants to have @enerbits
> <https://github.com/enerbits> code on ESP8266. and then a question.
> here is the code:
>
> /************************************************************** * * For this example, you need to install CRC32 library: *   https://github.com/bakercp/CRC32 *   or from http://librarymanager/all#CRC32+checksum * * TinyGSM Getting Started guide: *   http://tiny.cc/tiny-gsm-readme * * ATTENTION! Downloading big files requires of knowledge of * the TinyGSM internals and some modem specifics, * so this is for more experienced developers. * **************************************************************/
> // Select your modem:
> #define TINY_GSM_MODEM_SIM800// #define TINY_GSM_MODEM_SIM808// #define TINY_GSM_MODEM_SIM900// #define TINY_GSM_MODEM_A6// #define TINY_GSM_MODEM_A7// #define TINY_GSM_MODEM_M590// #define TINY_GSM_MODEM_ESP8266// #define TINY_GSM_MODEM_XBEE
> // Increase RX buffer if needed
> #define TINY_GSM_RX_BUFFER 1024
>
> #include <FS.h>
> #include <TinyGsmClient.h>
> #include <CRC32.h>
> // Uncomment this if you want to see all AT commands//#define DUMP_AT_COMMANDS
> // Set serial for debug console (to the Serial Monitor, default speed 115200)
> #define SerialMon Serial
> // Use Hardware Serial on Mega, Leonardo, Micro//#define SerialAT Serial1
> // or Software Serial on Uno, Nano
> #include <SoftwareSerial.h>
> SoftwareSerial SerialAT(D2, D1); // RX, TX
>
> // Your GPRS credentials// Leave empty, if missing user or passconst char apn[]  = "mtnirancell";  // replace with your apnconst char user[] = "";const char pass[] = "";
> [... cut ...]
> const char resource[]  = "/filelink/w93wgDpMMMKg/p6nv_gateway.bin"; // my program is a simple blink on LED_BUILTIN pin, so u can use it as welluint32_t knownCRC32    = 0x6f50d767;uint32_t knownFileSize = 1024;   // In case server does not send it
> [... 241 lines cut ...]
>
> now my question is: does the connection really have a time limit? how can
> I change it?
>
> and also if u used the code above, I'll be glad to know the result.

Thank you for your response. Yes the HTTP has connection time. Please refer
this article -
https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Keep-Alive.

And in my case I used wemos D1 R1 development board and SIM800 GSM Module,
the TinyGSM library is not working properly.  AT Commands are not passing
to the module. Any idea about it?

Black3rror, 2018-11-27:
> Thank you for your response. Yes the HTTP has connection time. Please refer this article - https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Keep-Alive. And in my case I used wemos D1 R1 [...]

Thanx for response. at first I had the same problem. I mean sometimes my SIM800 module got "AT" correctly but sometimes got "?T" instead of that. I've decreased the baud rate from 115200 to 9600 and used a PCB instead of wiring and the problem gone. hope decreasing the baud rate and soldering instead of wiring on bread board solve your problem.

Warlib1975, 2019-04-18:
Here is some description, how I used ESP32 with SIM800L http://www.bizkit.ru/en/2019/04/03/13540/ with HardwareSerial. Maybe will be useful. It's MT, I haven't corrected yet the quality of translation, sorry.

tarcisocallegario, 2019-06-03:
> @vshymanskyy
> Hello
> Yes, you are absolutely right, feed the updater, it would be great.
> Due to my little knowledge about the free updater, my direct feed attempts did not work. And what you saw is the only way I found to make it work. Totally inefficient by the way.
> 
> Well ... going back to the topic that brought us.
> At last I was able to make the client not disconnect after several hours of operation (sending GPS data) using sim800 ...
> 
> in my case the solution was simple.
> short cables and well welded. It's more ... now to make sure I've soldier what could be said ... pin with pin
> 
> now ... no more problems ...
> my suspicion started when I inspected the talk between the esp and the sim800 ...
> 
> the famous "?" they appeared from time to time, and that's when the disaster happened ...
> 
> then it appeared,
> my amateur radio spirit ... that has to be noise.
> 
> I quickly removed the cables and welded pin with pin
> 
> ready. in my case.
> the problem was solved
> 
> I hope it serves someone.
> 
> Thank you!!!!
> 
> Pablo

Hi, Pablo
    Have you done any upgrade in your OTAcode ?
Best Regards !

elyasssmart, 2021-02-20:
> @enerbits thanx for code sharing.
> I have a code for those who wants to have @enerbits code on ESP8266. and then a question.
> here is the code:
> 
> ```c++
> /**************************************************************
>  *
>  * For this example, you need to install CRC32 library:
>  *   https://github.com/bakercp/CRC32
>  *   or from http://librarymanager/all#CRC32+checksum
>  *
>  * TinyGSM Getting Started guide:
>  *   http://tiny.cc/tiny-gsm-readme
>  *
>  * ATTENTION! Downloading big files requires of knowledge of
>  * the TinyGSM internals and some modem specifics,
>  * so this is for more experienced developers.
>  *
>  **************************************************************/
> 
> // Select your modem:
> #define TINY_GSM_MODEM_SIM800
> // #define TINY_GSM_MODEM_SIM808
> // #define TINY_GSM_MODEM_SIM900
> // #define TINY_GSM_MODEM_A6
> // #define TINY_GSM_MODEM_A7
> // #define TINY_GSM_MODEM_M590
> // #define TINY_GSM_MODEM_ESP8266
> // #define TINY_GSM_MODEM_XBEE
> 
> // Increase RX buffer if needed
> #define TINY_GSM_RX_BUFFER 1024
> 
> #include <FS.h>
> #include <TinyGsmClient.h>
> #include <CRC32.h>
> 
> // Uncomment this if you want to see all AT commands
> //#define DUMP_AT_COMMANDS
> 
> // Set serial for debug console (to the Serial Monitor, default speed 115200)
> #define SerialMon Serial
> 
> // Use Hardware Serial on Mega, Leonardo, Micro
> //#define SerialAT Serial1
> 
> // or Software Serial on Uno, Nano
> #include <SoftwareSerial.h>
> SoftwareSerial SerialAT(D2, D1); // RX, TX
> 
> 
> // Your GPRS credentials
> // Leave empty, if missing user or pass
> const char apn[]  = "mtnirancell";  // replace with your apn
> const char user[] = "";
> const char pass[] = "";
> 
> // Server details
> const char server[] = "uupload.ir"; // replace with your server name
> const int  port = 80;
> 
> const char resource[]  = "/filelink/w93wgDpMMMKg/p6nv_gateway.bin"; // my program is a simple blink on LED_BUILTIN pin, so u can use it as well
> uint32_t knownCRC32    = 0x6f50d767;
> uint32_t knownFileSize = 1024;   // In case server does not send it
> 
> #ifdef DUMP_AT_COMMANDS
>   #include <StreamDebugger.h>
>   StreamDebugger debugger(SerialAT, SerialMon);
>   TinyGsm modem(debugger);
> #else
[... 237 lines cut ...]
>                 Serial.println("OTA didn't finish? something went wrong!");
>             }
>         }
>         else
>         {
>             Serial.println("Error occured #: " + String(Update.getError()));
>         }
>     }
>     else
>     {
>         Serial.println("without enough space to do OTA");
>     }
> }
> ```
> 
> I had problem downloading the file. for me it takes about 3 to 4 minutes to download it so after about 80% I think it exceeds the time limit of connection and disconnects from server. so I can't download it anyway.
> but I've tested it by uploading update.bin file to SPIFFS directly from my computer and then trying updateFromFS(). result: it worked. so probably complete code also works.
> 
> now my question is: does the connection really have a time limit? how can I change it?
> 
> and also if u used the code above, I'll be glad to know the result.

Hey Dear,
tnx for code
i use for My Wemos D1 and work..

rjjimenezl, 2023-07-13:
Hi, can someone help me with this problem I'm having? I'm using ESP32-WROOM-32U, SIM 808 EVB-V3.2 UART HardwareSerial communication I'm trying to send GPS data to an account in ubidots but I get this error, I've read it Almost everything and nothing has solved this problem for me, despite the fact that in the past I was able to run the same project without any problem, but now that I want to do it again, I can't because of this error. I would appreciate the help in advance.
[image]
that line that says:
getDateTimeStampGSM_modem()
It is the last function that it enters before throwing the error.

I attach the complete function:

long getDateTimeStampGSM_modem()
{
  //Obtém string com os dados de localização
  String loc = modem.getGsmLocation();
  Serial.println("getGsmLocation()");
  String dateTime;
  int timestamp;
  Serial.println("timestamp");
  //Obtém substring com a data e hora
  getDateTime_GSMBuffer(loc, &dateTime);     
  //Calcula o timestamp baseado na data/hora
  timestamp = timeStamp_GSMBuffer(dateTime);

  if(timestamp == -1)
    Serial.println("Failed to obtain timestamp (GSM): "+dateTime);   
    
  return timestamp;
}

lukwagohakim, 2024-07-21:
a7672e modem
esp wroom 32
first was failed to connect to vsh.pp.ua
now its failed to connect to the apn
`/**************************************************************
 *
 * This sketch connects to a website and downloads a page.
 * It can be used to perform HTTP/RESTful API calls.
 *
 * TinyGSM Getting Started guide:
 *   https://tiny.cc/tinygsm-readme
 *
 **************************************************************/

// Select your modem:
// #define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_SIM7000
// #define TINY_GSM_MODEM_SIM7000SSL
// #define TINY_GSM_MODEM_SIM7080
// #define TINY_GSM_MODEM_SIM5360
// #define TINY_GSM_MODEM_SIM7600
#define TINY_GSM_MODEM_A7672X
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_SARAR4
// #define TINY_GSM_MODEM_SARAR5
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
// #define TINY_GSM_MODEM_ESP32
// #define TINY_GSM_MODEM_XBEE
// #define TINY_GSM_MODEM_SEQUANS_MONARCH

#include <TinyGsmClient.h>

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
//#ifndef __AVR_ATmega328P__
//#define SerialAT Serial1

// or Software Serial on Uno, Nano
//#else
//#include <SoftwareSerial.h>
//SoftwareSerial SerialAT(2, 3);  // RX, TX
//#endif

#include <HardwareSerial.h>

HardwareSerial SerialAT(2);

// Define the DTR and RI pins used for hardware flow control
const int dtrPin = 18;
const int riPin = 19;

// Increase RX buffer to capture the entire response
// Chips without internal buffering (A6/A7, ESP8266, M590)
// need enough space in the buffer for the entire response
// else data will be lost (and the http library will fail).
#if !defined(TINY_GSM_RX_BUFFER)
[... 191 lines cut ...]
      timeout = millis();
    }
  }
  SerialMon.println();

  // Shutdown

  client.stop();
  SerialMon.println(F("Server disconnected"));

#if TINY_GSM_USE_WIFI
  modem.networkDisconnect();
  SerialMon.println(F("WiFi disconnected"));
#endif
#if TINY_GSM_USE_GPRS
  modem.gprsDisconnect();
  SerialMon.println(F("GPRS disconnected"));
#endif

  // Do nothing forevermore
  while (true) { delay(1000); }

  // Set the DTR pin LOW to signal the SIM7672S module to stop sending data
  digitalWrite(dtrPin, LOW);
}`