vshymanskyy/TinyGSM #135: Blynk and SMS and Dial Funcs OnTime

vkskif, 2018-02-26:
I try to use manage NodeMcu by SMSs, Calls-DTMF - and respond by SMS - it's work fine!
I try to represent data by BLYNK - it's fine too!
But when i try to concatenate code - SMS, CALLS,BLYNK - NodeMcu is STOPPED  - crashed with error
-------------------------
TEST SerialMon Output
First start app: Initializing modem...
AT

OK
AT+CFUN=0

+SAPBR 1: DEACT

+CPIN: NOT READY

OK
AT+CFUN=1,1

OK
AT
AT
AT

OK
AT&FZ
AT&FZ

OK
ATE0
ATE0

OK
AT+CPIN?

ERROR
AT+CPIN?

ERROR
AT+CPIN?

+CPIN: READY

OK
First start app: after delay 5000
AT+CPIN?

+CPIN: READYAT+CMGDA="DEL ALL"
AT+CMGDA="DEL ALL"

ERROR
AT+CMGDA="DEL ALL"

ERROR
AT+CPMS=ME,ME,ME

ERROR
AT+CMGF=1

OK
AT+CNMI=1,0

ERROR
AT+CLIP=1

OK
AT+CSCS="GSM"

ERROR
AT+DDET=1,1000,0,0

OK
AT

OK
AT&FZ

OK
ATE0
ATE0

OK
AT+CPIN?

+CPIN: READY

OK
AT+CPIN?

+CPIN: READY

OK
AT+CREG?

+CREG: 0,5

OK
AT+COPS?
AT+CIPSHUT

SHUT OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","3g.utel.ua"

OK
AT+SAPBR=3,1,"USER",""
AT+SAPBR=3,1,"PWD",""

OK
AT+CGDCONT=1,"IP","3g.utel.ua"

OK
AT+CGACT=1,1

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"46.203.235.203"

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

ERROR

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
--------------------------------------------------

**HOW TO USE BLYNK AND OTHER FUNCs sim800 MODULE???
IT'S POSSIBLE???**

hubaksis, 2018-02-26:
Please format your post using [markdown](https://guides.github.com/features/mastering-markdown/). Very difficult to read.

vkskif, 2018-03-01:
has't there people that used blynk and sms/dtmf ONTIME or try to do that???

hubaksis, 2018-03-01:
I don't think that this is an issue with TinyGSM library, so probably this is not a right place to look for help.
Anyway, "wdt reset" means that the wdt (watchdog) caused a reset. It means that the program is not responding. Debug your code and find when it crashes. Try to check your code for memory leaks (available memory after each steps), try to add a code piece by piece, and other debug stuff.

vkskif, 2018-03-04:
Solved, 
I delete Blynk, TinyGSM, Timelib - libraries manualy
Than install only one last Blynk (with TinyGSM and Timelib) and all features blynk and sms/dtmf simultaneously work now!
Thanks for help!

closed 2018-03-04 by vkskif