vshymanskyy/TinyGSM #136: Connection with AWS IoT

shirish47, 2018-03-01:
Hi, does TinyGSM support tls 1.2 over GSM ? 
I want to see the possibility of connecting to AWS IoT over GSM. But finding none.

hubaksis, 2018-03-01:
TinyGSM uses AT commands to talk to a GSM modem to connect to something [http, https, etc.] over a mobile network, If you modem supports the stuff you are asking and TinyGSM supports this modem, then it will work,

vshymanskyy, 2018-03-09:
@shirish47 I don't know of any ways of setting the Client Certificate.

daz, 2018-03-17:
Certificates and private keys can be set on U201, I managed to connect to AWS IoT on an MKRGSM1400 dev board but I just used the MKRGSM library and a few AT commands. It’s doable.

vshymanskyy, 2018-03-17:
@daz thanks for your input. could you provide an example sketch? I think we'll be able to incorporate that :)

daz, 2018-03-17:
I haven’t got a code example right now sorry but here’s at AT commands:

```
Import certs
>AT+USECMNG=0,0,"ROOT_CERT",1706
>[Send cert string, length 1706]
+USECMNG: 0,0,"ROOT_CERT","cb17e4..."
OK

>AT+USECMNG=0,1,"CLIENT_CERT",1206
>[Send cert string, length 1206]
+USECMNG: 0,1,"CLIENT_CERT","081dd1..."
OK

>AT+USECMNG=0,2,"PRIVATE_KEY",1654
>[Send cert string, length 1654]
+USECMNG: 0,2,"PRIVATE_KEY","bcb22b..."
OK

Profile 1, cert validation level 1
AT+USECPRF=1,0,1
OK

Profile 1, server can use any version for the connection
AT+USECPRF=1,1,0
OK

Profile 1, use named root cert
AT+USECPRF=1,3,"ROOT_CERT"
OK

Profile 1, use named client cert
AT+USECPRF=1,5,"CLIENT_CERT"
OK

Profile 1, use named private key
AT+USECPRF=1,6,"PRIVATE_KEY"
OK

Create TCP socket
AT+USOCR=6
+USOCR: 0
OK

Previous returned socket 0, using SSL/TLS, apply profile 1
AT+USOSEC=0,1,1
OK
```

daz, 2018-04-11:
Here's my fork of MKRGSM https://github.com/daz/MKRGSM

There's an example for connecting to AWS IoT too:
https://github.com/daz/MKRGSM/blob/master/examples/GsmSecurity/GsmSecurity.ino

Happy to help port it to TinyGSM, my C++ knowledge is extremely limited though

closed 2018-04-23 by vshymanskyy

ankitmhn, 2018-06-24:
Has anyone been able to connect to AWS IoT using Sim808? I had the Arduino device connect to PubNub using MQTT using unsecured port 1883. AWS IoT requires 3 certificates from each device that connects to it. Anyone has any solution?

jonnor, 2018-09-26:
Is it possible when using a device that has its own TLS implementation (ESP8266/ESP32) to let it handle the TLS parts? Or must it always be done by the GSM module?

jonnor, 2018-09-26:
Yeah, looks like PPP/PPPoS (over serial) is the keyword. There is an example for the esp-idk here: https://github.com/loboris/ESP32-PPPOS-EXAMPLE

Thanks for the help!

vshymanskyy, 2018-10-15:
@jonnor, did you have any success with this? Could you share your findings?

jonnor, 2018-10-15:
I have not had time to test yet. Will share here when I have!

gilbertgit, 2018-10-29:
> I haven’t got a code example right now sorry but here’s at AT commands up until applying the TLS profile to the socket. From there all I did was change the socket connect AT command in MKRGSM library’s SSL connect so it used that TLS profile too. I’m hoping to patch their library in the coming days, I’d love to help getting it working in TinyGSM too.
> 
> ```
> Import certs
> >AT+USECMNG=0,0,"ROOT_CERT",1706
> >[Send cert string, length 1706]
> +USECMNG: 0,0,"ROOT_CERT","cb17e4..."
> OK
> 
> >AT+USECMNG=0,1,"CLIENT_CERT",1206
> >[Send cert string, length 1206]
> +USECMNG: 0,1,"CLIENT_CERT","081dd1..."
> OK
> 
> >AT+USECMNG=0,2,"PRIVATE_KEY",1654
> >[Send cert string, length 1654]
> +USECMNG: 0,2,"PRIVATE_KEY","bcb22b..."
> OK
> 
> Profile 1, cert validation level 1
> AT+USECPRF=1,0,1
> OK
> 
> Profile 1, server can use any version for the connection
> AT+USECPRF=1,1,0
> OK
> 
> Profile 1, use named root cert
> AT+USECPRF=1,3,"ROOT_CERT"
> OK
> 
> Profile 1, use named client cert
> AT+USECPRF=1,5,"CLIENT_CERT"
> OK
> 
> Profile 1, use named private key
> AT+USECPRF=1,6,"PRIVATE_KEY"
> OK
> 
> Create TCP socket
> AT+USOCR=6
> +USOCR: 0
> OK
> 
> Previous returned socket 0, using SSL/TLS, apply profile 1
> AT+USOSEC=0,1,1
> OK
> ```

> Here's my fork of MKRGSM https://github.com/daz/MKRGSM
> 
> There's an example for connecting to AWS IoT too:
> https://github.com/daz/MKRGSM/blob/master/examples/GsmSecurity/GsmSecurity.ino
> 
> Happy to help port it to TinyGSM, my C++ knowledge is extremely limited though

Why do you have a +2 on the size of the certs being sent to the modem?

gilbertgit, 2018-10-29:
> Here's my fork of MKRGSM https://github.com/daz/MKRGSM
> 
> There's an example for connecting to AWS IoT too:
> https://github.com/daz/MKRGSM/blob/master/examples/GsmSecurity/GsmSecurity.ino
> 
> Happy to help port it to TinyGSM, my C++ knowledge is extremely limited though

I'm trying to implement your branch and I'm not having any luck. Did you have to generate your own certs or did you use the ones that aws iot generated for you? Also why do you have a + 2 on the size for the certs being sent up?

rreddys, 2019-04-19:
Here's my fork of MKRGSM https://github.com/daz/MKRGSM

There's an example for connecting to AWS IoT too:
https://github.com/daz/MKRGSM/blob/master/examples/GsmSecurity/GsmSecurity.ino

Happy to help port it to TinyGSM, my C++ knowledge is extremely limited though

can you help me Daz am facing issues to connect to AWS with your code .

daz, 2019-04-19:
Happy to help, hit me on Twitter @dazonic

AhmadSalmanKhan, 2019-07-15:
Hi guys!
Any update on that TinyGSM AWS issue? Where are you guys with the TLS 1.2 port to tiny gsm, I'll be happy to take part in the port.

Moreover, will the port work for modems like sim808 that support standard SSL or TLS 1.0 but not the latest one's like TLS1.2 ?
Or Does the port needs the modem to support the standard in order to work ?

Because from my understanding TLS 1.2 is implemented in software, so if we can create a client implementation of TLS 1.2 compatible with any Gsm modem, we can use it on whatever modem we like?

Please tell me if this approach is plain wrong.