vshymanskyy/TinyGSM #139: GPS accuracy 2 digits (float) not enogh

BullSimons, 2018-03-07:
Hello,

i need higher accuracy than only 2 digits. Is it possible to get lat/lon as a long? Or does the SIM808 only Support "short" latitude/longitude?

Thanks!

hubaksis, 2018-03-07:
`AT+CGNSINF` returns for SIM808 in the parameters 4 and 5 ([source, page 10](https://www.elecrow.com/download/SIM800%20Series_GNSS_Application%20Note%20V1.00.pdf))
`Latitude format: ±dd.dddddd; range: [-90.000000,90.000000]; length: 10`
`Longitude format: ±ddd.dddddd; rnage: [-180.000000,180.000000]; length 11`

Please specify where exactly you get 2 digits only.

BullSimons, 2018-03-07:
Thank you for the Information!
In every library which i looked at for latitude and longitude are declared as "float" so you only get 2 digits. 
As you showed must it be possible to get more accuracy - i just need to replace float by long.

But my programming skills are bad (such as my englisch ;)   ) 
I will try my very best...

hubaksis, 2018-03-07:
Actually, float has 7 digits precision ([source](https://stackoverflow.com/questions/2386772/difference-between-float-and-double)).
Long doesn't have decimal points at all, it's integer.

brvdg, 2018-03-07:
Hi, why do you want long instead of float? Both has 32Bit.

> Long variables are extended size variables for number storage, and store 32 bits (4 bytes), from -2,147,483,648 to 2,147,483,647.
[https://www.arduino.cc/reference/en/language/variables/data-types/long/](Arduino reference for long)

> Datatype for floating-point numbers, a number that has a decimal point. ... Floating-point numbers can be as large as 3.4028235E+38 and as low as -3.4028235E+38. They are stored as 32 bits (4 bytes) of information.
[https://www.arduino.cc/reference/en/language/variables/data-types/float/](Arduino reference for float)

BullSimons, 2018-03-07:
You are right. 
In another project i used a ublox gps. There latitude and longitude are declared as Long - which works with that other library. 

But in Arduino float has only 2 Digits because it is only 4byte?

Any suggestions?

brvdg, 2018-03-07:
> But in Arduino float has only 2 Digits because it is only 4byte?
Are you sure?

long and float has the same size (32Bit).
But long is not usable for GPS. Because long can only handle full numbers and no floating points.

BullSimons, 2018-03-07:
Solved.
I thought  48.699364 will be converted to 48.70 an all the other Digits will be ereased. 
But -> 48.699364 * 1000000 -> 48699364 (because of that in the other library it is a Long)

Damn - i hate programming ;)

closed 2018-03-07 by BullSimons

brvdg, 2018-03-07:
I think I understand your problem.
You using a function which round the float to two digits.
e.g. Serial.print(float, 2)
use Serial.print(float, 5) and you see five digits. Or more if you want.

Which library use long for GPS?