vshymanskyy/TinyGSM #140: Recive ussd, send sms problem... Help

spa-sam, 2018-03-09:
sim800
1. When I send a request to "# 101 #" or "* 161 #", I get an empty response. 
2. When sending SMS on the phone I get a set of different characters instead of "Hello world!".

At the factory settings reset - does not help!

vshymanskyy, 2018-03-09:
Could you play with your modem with AT commands, and tell me what sequences work for you?

spa-sam, 2018-03-09:
With sms found out - you had to add the translation into the  "gsm" encoding:
bool sendSMS(const String& number, const String& text) {
    sendAT(GF("+CMGF=1"));
    waitResponse();
    sendAT(GF("+CSCS=\"GSM\""));                              //add
    waitResponse();
    sendAT(GF("+CMGS=\""), number, GF("\""));
...
It remains to defeat ussd ...

spa-sam, 2018-03-09:
USSD

AT+CMGF=1
OK
AT+CSCS="HEX"
OK
AT+CUSD=1,"*101#"
OK
+CUSD: 2

hubaksis, 2018-03-09:
This issue is with the alphabet and it was discussed [here](https://github.com/vshymanskyy/TinyGSM/issues/2#issuecomment-362951233)
Pending [pull request](https://github.com/vshymanskyy/TinyGSM/pull/128)

vshymanskyy, 2018-03-09:
@hubaksis indeed, looks related. I'll take a look asap

vshymanskyy, 2018-03-14:
@spa-sam could  you verify on master branch?

spa-sam, 2018-03-17:
Ussd works only like this:
  String sendUSSD(const String& code) {
    sendAT(GF("+CMGF=1"));
    waitResponse();
    sendAT(GF("+CSCS=\"GSM\""));  //     GSM!
    waitResponse();
    sendAT(GF("+CUSD=1,\""), code, GF("\""));
    if (waitResponse() != 1) {
      return "";
    }
    if (waitResponse(10000L, GF(GSM_NL "+CUSD:")) != 1) {
      return "";
    }
    stream.readStringUntil('"');
    String hex = stream.readStringUntil('"');
    stream.readStringUntil(',');
    int dcs = stream.readStringUntil('\n').toInt();
	return hex;
   /* if (dcs == 15) {
      return TinyGsmDecodeHex8bit(hex);
    } else if (dcs == 72) {
      return TinyGsmDecodeHex16bit(hex);
    } else {
      return hex;
    }*/
  }

kamrulroky, 2018-11-01:
@spa-sam  I am facing similar issue, everything works with AT commands as yours but the changes you made in sendUSSD function also didn't work for me. Even with "HEX" instead of "GSM" gives me the phone number but nothing with "GSM".In addition blocking these condition made things worse,now I dont get location,time and date too.Any idea what might cause the difference? I am using sim800l

spa-sam, 2018-11-09:
It seems to me, it depends on the specific carrier.

coriolanweihrauch, 2019-03-27:
Facing the same issue and spa-sam's solution works for me. I've modified my library, but it would be great if this was integrated into the master source

Could you maybe integrate it into the source code with an additional parameter:
```
sendUSSD(const String& code, bool usedcs=true){
(...)
if( ! usedcs) {
    return hex;
}
else{
    if (dcs == 15) {
      return TinyGsmDecodeHex8bit(hex);
    } else if (dcs == 72) {
      return TinyGsmDecodeHex16bit(hex);
    } else {
      return hex;
    }
}
}
```
so that it is usable for everyone ?

marutichintan, 2021-03-23:
modem.sendSMS_UTF16("+380679837464", u"Привееет!", 10);

using above code i am able to send the unicode msg. but i want to cast string or char to UTF16 (char16_t). need help.