vshymanskyy/TinyGSM #143: use mqtt gprs and gps in same time

sbh96, 2018-03-20:
Hi 
I use sim808 GPRS with mqttClint_tiny gsm library on nodemcu.
when i use getGPS() connection between mqtt server and nodemcu lost and modem reconnect to the server.
how can i use GPS with GPRS without losing connection?

himanshu-hearthacker, 2018-03-20:
Try to increase the keep alive time in pubsubclient.h. May your keepalive time is very short.

DjordjeMandic, 2018-03-25:
@sbh96 by using gps you have to close gprs connection because data is sent via same serial interface, its impossible to keep gprs connection and send gps data at same time. you can get gps and then recconect back on mqtt server but for realtime transfer its impossible, thats why i had to buy a gps module. Maybe in future sim808 firmware will be opensource(i hope) so that we can mod it to send gps data on another serial port, as far as i know i think that sim808 has 3 serial ports but im not sure. So basically you cant use GPS and GPRS at same time because gps will interfere with gprs data on same serial port.

closed 2018-04-23 by vshymanskyy

ishdemon, 2021-04-12:
is it same for SIM7000 too? I mean i tried using both GPS and GPRS in SIM7000G but it closes the connection when i do `modem.getGPS`.