vshymanskyy/TinyGSM #145: code stuck  function

himanshu-hearthacker, 2018-03-21:
My code blocks infinitely  after 4-5 hours of running loop() function of Arduino. I am updating signal strength every 30 second but after 4-5 hours of operation. In debug using Serial.print() statements i found that my code using  my code Stuck  before the  waitResponse() function.
I am using SIM 808 for GPRS/ GPS data. I am sending GPS data to server every 10th second and updating Signal strength every 30 second.  
This  case generally happens when I am not getting any ACK from past 30-40 seconds. I am checking if my device is registered with network or not using getRegistrationStatus() method before updating signal strength.
I have attach two jpg files which shows status of code. Hoping to get some useful suggestion here.
([](https://drive.google.com/open?id=1aNNjYFei7uspV2l2yGldMPjG-ls8r6AH))
https://drive.google.com/open?id=1MxLp_4GZRWNm6XdAv4n8NqwO1lRL4eww

vshymanskyy, 2018-03-24:
it looks like your modem stopped responding. This may be due to power issue, unreliable modem serial connection, interference, etc.
You should definitely try enabling AT commands output to your serial to see what the actual modem response is.

vshymanskyy, 2018-04-23:
Was the issue resolved?

closed 2018-04-23 by vshymanskyy