vshymanskyy/TinyGSM #146: Library doesn't work on ESP32 but AT_Debug works perfectly

DjordjeMandic, 2018-03-23:
I tried to use library on ESP32 but it didnt work, so i went into atdebug and it works, i can make calls everything.... Then i switched back to examples from library and it doesnt work, it just doesnt receive data right. So here is a example log:

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Initializing modem...AT
AT

OK
AT+CFUN=0
⸮⸮Z fail
************************
 Is your modem connected properly?
 Is your serial speed (baud rate) correct?
 Is your modem powered on?
 Do you use a good, stable power source?
 Try useing File -> Examples -> TinyGSM -> tools -> AT_Debug to find correct configuration
************************
Waiting for network...AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
"⸮g*AT+CREG?
⸮;⸮⸮C7⸮⸮⸮⸮⸮af⸮⸮H⸮ݫ⸮T⸮N⸮1⸮AT+CREG?
H%⸮pAT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
⸮⸮⸮⸮⸮⸮⸮MX⸮І⸮r.u7Ԑ⸮⸮<⸮)S⸮DAT+CREG?
⸮Z⸮⸮AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
 fail
************************
 Is your sim card locked?
 Do you have a good signal?
 Is antenna attached?
 Does the SIM card work with your phone?
************************
Initializing modem...AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT

⸮^⸮⸮!⸮⸮AT
矃AT
⸮AT
R1AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
 fail
************************
 Is your modem connected properly?
 Is your serial speed (baud rate) correct?
 Is your modem powered on?
 Do you use a good, stable power source?
 Try useing File -> Examples -> TinyGSM -> tools -> AT_Debug to find correct configuration
************************
Waiting for network...AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
?.⸮⸮΍,ҍ⸮⸮⸮⸮W⸮QY⸮⸮_ATAT+CREG?
⸮⸮REAT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
AT+CREG?
G?

+CREG: 0,1

OK
 OK
Connecting to YourAPNAT+CIPSHUT
⸮⸮af⸮⸮H⸮AT+SAPBR=3,1,"Contype","GPRS"
AT+SAPBR=3,1,"APN","YourAPN"
AT+SAPBR=3,1,"USER",""
AT+SAPBR=3,1,"PWD",""
⸮⸮AT+CGDCONT=1,"IP","YourAPN"
` (i ignored apn because it meaningless when it fails to get data back, im planning to use mqtt)

Main problem: Data is received right when testing with AT_Debug but when using library or example sketches it doesnt work. ESP32 buffer problem?

DjordjeMandic, 2018-03-23:
I also cant use factory reset because it waits for return true on modem.init xD... I will try to skip taht part but idk if its smart to do it.

DjordjeMandic, 2018-03-23:
AT&FZE0&W
AT&FZE0&W

OK
AT+IPR=0

OK
AT+IFC=0,0
AT+ICF=3,3
AT+CSCLK=0
AT&W
***********************************************************
 Return settings to Factory Defaults: FAIL
***********************************************************
RIP... ohh im soo ruined, 1 month working on a project for school and it dosent work in end. main thing... I NEED THIS TO WORK TOMORROW ASHJDASDA
Im sorry for being annoying but im mad.

DjordjeMandic, 2018-03-23:
AT+ICF=3,3 gives me error

DjordjeMandic, 2018-03-23:
Fixed, looked at signals on oscilloscope and saw that they are full of noise, wires were next to buck converter....

closed 2018-03-23 by DjordjeMandic

vshymanskyy, 2018-03-24:
@DjordjeMandic cool. Nice catch! I'll add this to the doc ;)

ayavilevich, 2018-12-11:
I want to document a similar case where "AT_Debug" works but "Diagnostic" doesn't on a ESP32. In my case it was trying to initialize the modem with "AT" commands without success.

Checking with the logic analyzer I saw that garbage is being sent instead of "AT",0x0D,0x0A.
Looking in the code I saw that the library uses lots of templates and variable argument "Args" notations in the code that sends the data. 
I went to update my Arduino-ESP32 code and tools which resolved the issue. (https://github.com/espressif/arduino-esp32/)
I guess that at some point in time there were bugs in the ESP32 tool chain and it wouldn't build a valid program from the code.