vshymanskyy/TinyGSM #148: A7 hanging on http.get(resource) using HttpClient Example

roja, 2018-04-03:
TinyGSM version: 0.3.2
Hardware: ESP8266 + A7

### Scenario, steps to reproduce
Using the example "HttpClient" using the A7 over ports 14,12 and the o2 network in the uk

### Expected result
Should download logo.txt sucessfully

### Actual result
Hangs when http.get(resource) is called.

vshymanskyy, 2018-04-05:
Could you please provide the AT command log?

AmrAlaa-exe, 2018-04-09:
Same issue,
------------
Performing HTTP GET request... AT+CIPCLOSE=255

+CME ERROR:53
AT+CIPSTART="TCP","156.209.63.76",2000

+CIPNUM:0

CONNECT OK

OK
AT+CIPSEND=0,3

>GET 
OK
AT+CIPSEND=0,1

>  
OK
AT+CIPSEND=0,7

>/api/a7
OK
AT+CIPSEND=0,9

> HTTP/1.1 
OK
AT+CIPSEND=0,2

>
 
OK
AT+CIPSEND=0,6

>Host:  
OK
AT+CIPSEND=0,13

>156.209.63.76 
OK
AT+CIPSEND=0,1

>: 
OK
AT+CIPSEND=0,4

>2000 
OK
AT+CIPSEND=0,2

>
 
OK
AT+CIPSEND=0,10

>User-Agent 
OK
AT+CIPSEND=0,2

>:  
OK
AT+CIPSEND=0,13

>Arduino/2.2.0 AT+CIPSEND=0,2

>
 AT+CIPSEND=0,10
AT+CIPSEND=0,2

>:  AT+CIPSEND=0,5
AT+CIPSEND=0,2

>
 AT+CIPSEND=0,2

> 
+TCPCLOSED:0

+CME ERROR:53
-3

vshymanskyy, 2018-04-10:
@AmrAlaa-exe what hardware do you use?

AmrAlaa-exe, 2018-04-12:
@vshymanskyy  ATmega32U4 combined with the A7 module.
https://goo.gl/SHVG9m

vshymanskyy, 2018-04-23:
@roja any updates?

vshymanskyy, 2018-04-23:
Closing this issue as no additional info provided

closed 2018-04-23 by vshymanskyy