vshymanskyy/TinyGSM #152: OTA using SIM800

thbl, 2018-04-13:
Hi Guys
Does any one have an example on how to do a OTA over sim800 and arduino ?
-Or how/where can i get startet with it ? :)

hubaksis, 2018-04-14:
I have successfully made OTA for ESP8266. In general:
1. Send a request to the server.
2. Read raw response to some kind of memory using read method 
```
uint8_t file_buffer[512];

 while (client->available()){
int read = client->read(file_buffer, sizeof(file_buffer) - 1);
//save file_buffer somewhere
}
```
3. Update.

So, you need to find out if it's possible to make OTA for arduino and how it works. There is no problems downloading a file using TinyGSM.

closed 2018-04-14 by vshymanskyy

kamrulroky, 2018-04-15:
@hubaksis after downloading the bin file where did you save the file and how do you upload the file from there. I am really a rookie to this, trying OTA for esp8266 using SIM800. can you share some details how its done, please

hubaksis, 2018-04-15:
File is saved to the SPIFFS ([docs here](http://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#filesystem-information-structure))
Then you need to perform an update from SPIFFS.

lucasromeiro, 2022-01-27:
Hi guys, I have a problem.

I use esp32 and a sim800l modem.
I need to do OTA update safely, so there is no risk of a device breaking with a failed update.
However, apparently the tinygsm communication object is not supported for the standard OTA update function.
I have a large program, almost 1mb of code.
I looked at people downloading the BIN file for SPIFFS and then updating. Is this 100% safe? Is there a risk of failure and killing equipment that is remote?
I tried to implement it but I couldn't because the . BIN is too big.

Maybe if I change the file system to 1M of program and 3M of SPIFFS. It is possible?

If there is another way, can you help me?

mudgalp, 2022-04-19:
i'm also facing same problem as you are facing @lucasromeiro 
please help anyone.

Thank you

gotnull, 2023-01-14:
And update on this? (Pun not intended).

jfbvm, 2025-03-27:
Hi to everyone,
I had the same question, and it was very difficult to find a code solution, so if anyone yet has this doubt, please take a look at the code below.
I'm already using the code below with [vshymanskyy/TinyGSM](https://github.com/vshymanskyy/TinyGSM) and it is working perfectly, the only problem is the download speed, this library doesn't have a file download optimization, so the speed will be veeeery low (max 14 or 20 KBps), but it worked!
I'm using an ESP32-S3 with 16MB of internal flash and PSRAM

```cpp
const char OTA_server[] = "YOUR_SERVER_TCP_URL";
const int OTA_server_port = 80;
const char OTA_bin_filename[] = "/YOUR_BIN_FILENAME.bin";
uint32_t OTA_knownCRC32 = 0x6f50d767;
uint32_t OTA_knownFileSize = 1024;  // In case server does not send it
bool OTA_UPDATING = false;

void appendFile(fs::FS& fs, const char* path, const char* message) {
    Serial.printf("[*] Appending to file: %s\n", path);

    File file = fs.open(path, FILE_APPEND);
    if (!file) {
        Serial.println("[-] Failed to open file for appending");
        return;
    }
    if (file.print(message)) {
        Serial.println("APOK");
    } else {
        Serial.println("APX");
    }
}
void readFile(fs::FS& fs, const char* path) {
    Serial.printf("[*] Reading file: %s\n", path);

    File file = fs.open(path);
    if (!file || file.isDirectory()) {
        Serial.println("[-] Failed to open file for reading");
        return;
    }

    Serial.print("[*] Read from file: ");
    while (file.available()) {
        Serial.write(file.read());
        delayMicroseconds(100);
    }
}
void writeFile(fs::FS& fs, const char* path, const char* message) {
    Serial.printf("[*] Writing file: %s\n", path);

    File file = fs.open(path, FILE_WRITE);
    if (!file) {
        Serial.println("[-] Failed to open file for writing");
        return;
    }
    if (file.print(message)) {
        Serial.println("[+] File written");
    } else {
        Serial.println("[-] Write failed");
    }
}
void listDir(fs::FS& fs, const char* dirname, uint8_t levels) {
    Serial.printf("[*] Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if (!root) {
        Serial.println("[-] Failed to open directory");
        return;
    }
    if (!root.isDirectory()) {
        Serial.println("[-] Not a directory");
        return;
    }

    File file = root.openNextFile();
[... 188 lines cut ...]
      Serial.println(contentLength);
      Serial.print("wrote:  ");
      Serial.println(readLength);
      Serial.print("Calculado. CRC32:    0x");
      Serial.println(crc.finalize(), HEX);
      Serial.print("Conocido CRC32:    0x");
      Serial.println(OTA_knownCRC32, HEX);
      Serial.print("Bajado en:       ");
      Serial.print(duration);
      Serial.println("s");

      Serial.println("wait fot 3 seconds");
      for (int i = 0; i < 3; i++)
      {
          Serial.print(String(i) + "...");
          delay(1000);
      }
    */
    //readFile(FileFS, "/update.bin");

    updateFromFS();

    return true;
}
```