vshymanskyy/TinyGSM #157: Quectel BG96 LTE modem support

vshymanskyy, 2018-04-23:
I have just added (a very early) support of **Quectel BG96**:
https://github.com/vshymanskyy/TinyGSM/blob/master/src/TinyGsmClientBG96.h

Currently, only some basic stuff is implemented: initialization, gprsConnect, TCP connection.
Diagnostics sketch looks working, didn't try other protocols yet.

You can contribute to this development by:
- Testing it
- Fixing non-working functionality
- Reviewing implementation to check if it works according to Quectel manuals:
[Quectel_BG96_AT_Commands_Manual_V2_0](https://github.com/vshymanskyy/TinyGSM/blob/master/extras/doc/Quectel_BG96_AT_Commands_Manual_V2_0.pdf)
[Quectel_BG96_TCPIP_AT_Commands_Manual_V1.0](https://github.com/vshymanskyy/TinyGSM/blob/master/extras/doc/Quectel_BG96_TCPIP_AT_Commands_Manual_V1.0.pdf)

Thanks ;)

gjt211, 2018-06-24:
Hi, I have tested this with a Quectel EC25 ad so far the MQTT test is working ok.
I will perform further testing in the coming weeks and will most likely create a new file specifically for the EC20/21/25 modems.

ThePrefixCyber, 2019-07-15:
I am experimenting with Quectel BC66 and BG96 and MQTT right now.  Initial tests don't work.  Are either of these models implemented with MQTT right now, or within range for me to make them work with simple modifications?

SRGDamia1, 2019-07-15:
The BG96 should work.  Can you post an AT log?

ThePrefixCyber, 2019-07-15:
I'm using the Dragino NB-IoT Shield with an Arduino Uno, software serial pins on pins 10, 11 and baud rate is 38400 for both serial monitor and software serial.  Looking through the log, I don't see any evidence that any MQTT related commands are being sent.  

```

Wait...
17:13:30.563 -> Initializing modem...
17:13:30.563 -> AT
17:13:30.563 -> AT

17:13:30.563 -> OK
17:13:30.563 -> AT+CFUN=1,1
17:13:30.563 -> AT+CFUN=1,1

17:13:30.600 -> OK
17:13:30.600 -> 
17:13:30.600 -> POWERED DOWN[6036] ### TinyGSM Version: 0.7.9
17:13:33.602 -> AT
17:13:33.602 -> 
17:13:33.602 -> 
17:13:33.602 -> +QIURC: "pdpdeact",1
17:13:33.895 -> AT
17:13:34.184 -> AT
17:13:34.512 -> AT
17:13:34.802 -> AT
17:13:35.128 -> AT
17:13:35.411 -> AT
17:13:35.736 -> AT
17:13:36.027 -> AT
17:13:36.343 -> AT
17:13:36.489 -> 
17:13:36.489 -> RDY
17:13:36.526 -> [8972] ### Unhandled: RDY
17:13:36.632 -> AT
17:13:36.632 -> AT

17:13:36.632 -> OK
17:13:36.632 -> AT&FZE0
17:13:36.632 -> AT&FZE0

17:13:36.667 -> OK
17:13:36.667 -> [9091] ### Modem: Quectel BG96
17:13:36.667 -> AT+CPIN?
17:13:36.667 -> 
17:13:36.667 -> +CPIN: READY
17:13:36.667 -> 
17:13:36.667 -> OK
17:13:36.667 -> ATI
17:13:36.667 -> 
17:13:36.667 -> Quectel
17:13:36.704 -> BG96
17:13:36.704 -> Revision: BG96MAR02A07M1G
17:13:36.704 -> 
17:13:36.704 -> OK
17:13:36.704 -> Modem: Quectel BG96 Revision: BG96MAR02A07M1G
17:13:36.704 -> Waiting for network...AT+CREG?
17:13:36.704 -> 
17:13:36.704 -> +CREG: 0,4
17:13:36.704 -> 
17:13:36.704 -> OK
17:13:36.967 -> AT+CREG?
17:13:36.967 -> 
17:13:36.967 -> +CREG: 0,2
17:13:36.967 -> 
17:13:36.967 -> OK
17:13:37.220 -> AT+CREG?
17:13:37.220 -> 
17:13:37.220 -> +CREG: 0,2
17:13:37.257 -> 
17:13:37.257 -> OK
17:13:37.477 -> AT+CREG?
17:13:37.512 -> 
17:13:37.512 -> +CREG: 0,2
17:13:37.512 -> 
17:13:37.512 -> OK
17:13:37.738 -> AT+CREG?
17:13:37.774 -> 
17:13:37.774 -> +CREG: 0,2
17:13:37.774 -> 
17:13:37.774 -> OK
17:13:38.025 -> AT+CREG?
17:13:38.025 -> 
17:13:38.025 -> +CREG: 0,2
17:13:38.025 -> 
17:13:38.025 -> OK
17:13:38.283 -> AT+CREG?
17:13:38.283 -> 
17:13:38.283 -> +CREG: 0,2
17:13:38.283 -> 
17:13:38.283 -> OK
17:13:38.535 -> AT+CREG?
17:13:38.569 -> 
17:13:38.569 -> +CREG: 0,2
17:13:38.569 -> 
17:13:38.569 -> OK
17:13:38.819 -> AT+CREG?
17:13:38.819 -> 
17:13:38.819 -> +CREG: 0,2
17:13:38.819 -> 
17:13:38.819 -> OK
17:13:39.074 -> AT+CREG?
17:13:39.074 -> 
17:13:39.074 -> +CREG: 0,2
17:13:39.074 -> 
17:13:39.074 -> OK
17:13:39.328 -> AT+CREG?
17:13:39.365 -> 
17:13:39.365 -> +CREG: 0,2
17:13:39.365 -> 
17:13:39.365 -> OK
17:13:39.608 -> AT+CREG?
17:13:39.608 -> 
17:13:39.608 -> +CREG: 0,2
17:13:39.608 -> 
17:13:39.608 -> OK
17:13:39.857 -> AT+CREG?
17:13:39.894 -> 
17:13:39.894 -> +CREG: 0,2
17:13:39.894 -> 
17:13:39.894 -> OK
17:13:40.147 -> AT+CREG?
17:13:40.147 -> 
17:13:40.147 -> +CREG: 0,1
17:13:40.147 -> 
17:13:40.147 -> OK
17:13:40.147 ->  OK
17:13:40.147 -> AT+CREG?
17:13:40.147 -> 
17:13:40.147 -> +CREG: 0,1
17:13:40.147 -> 
17:13:40.147 -> OK
17:13:40.147 -> Network connected
17:13:40.147 -> AT+QIRD=1,0
17:13:40.185 -> 
17:13:40.185 -> ERROR
17:13:40.185 -> AT+QISTATE=1,1
17:13:40.185 -> 
17:13:40.185 -> OK
17:13:41.167 -> [13611] ### Unhandled: OK
17:13:48.190 -> === MQTT NOT CONNECTED ===
17:13:48.190 -> Connecting to test.mosquitto.orgAT+QIRD=1,0
17:13:48.226 -> 
17:13:48.226 -> ERROR
17:13:48.226 -> AT+QISTATE=1,1
17:13:48.226 -> 
17:13:48.226 -> OK
17:13:49.219 -> [21632] ### Unhandled: OK
17:13:56.222 -> AT+QICLOSE=1
17:13:56.222 -> 
17:13:56.222 -> OK
17:13:56.259 -> AT+QIOPEN=1,1,"TCP","test.mosquitto.org",1883,0,0
17:13:56.259 -> 
17:13:56.259 -> OK
17:13:56.296 -> 
17:13:56.296 -> +QIOPEN: 1,565
17:13:56.296 ->  fail
17:13:56.367 -> AT+QIRD=1,0
17:13:56.404 -> 
17:13:56.404 -> ERROR
17:13:56.404 -> AT+QISTATE=1,1
17:13:56.404 -> 
17:13:56.404 -> OK
17:13:57.388 -> [29809] ### Unhandled: OK
17:14:04.432 -> === MQTT NOT CONNECTED ===
17:14:04.432 -> Connecting to test.mosquitto.orgAT+QIRD=1,0
17:14:04.432 -> 
17:14:04.432 -> ERROR
17:14:04.432 -> AT+QISTATE=1,1
17:14:04.470 -> 
17:14:04.470 -> OK
17:14:05.433 -> [37837] ### Unhandled: OK
17:14:12.469 -> AT+QICLOSE=1
17:14:12.469 -> 
17:14:12.469 -> OK
17:14:12.469 -> AT+QIOPEN=1,1,"TCP","test.mosquitto.org",1883,0,0
17:14:12.504 -> 
17:14:12.504 -> OK
17:14:12.504 -> 
17:14:12.504 -> +QIOPEN: 1,565
17:14:12.504 ->  fail
17:14:12.615 -> AT+QIRD=1,0
17:14:12.615 -> 
17:14:12.615 -> ERROR
17:14:12.615 -> AT+QISTATE=1,1
17:14:12.615 -> 
17:14:12.615 -> OK
17:14:13.649 -> [46010] ### Unhandled: OK

```

SRGDamia1, 2019-07-15:
This library doesn't use any MQTT specific commands for the BG96 (or any module) it only opens the TCP client.  Use the PubSubClient library (or other MQTT library of your choice) for actual MQTT communication formatting.  If you want to use the BG96's "built in" AT commands for MQTT, you'll have to do that on your own.

It looks like you're using something based off the Mqtt example, but not that example exactly.  You skipped the gprs connection step, which is why your TCP connection is failing.  That's lines 177 to 184 in the MQTT example.  Also make sure your APN is set correctly.

treyconley, 2020-03-03:
What BG96 boards are you guys using to test this. I can’t seem to find a good breakout board.

ThePrefixCyber, 2020-03-03:
Quectel will give you an eval board. Or sell you one if you’re not important enough.

SRGDamia1, 2020-03-03:
I've just recently acquired a Nimbelink Skywire (https://nimbelink.com/products/4g-lte-m-global-quectel/) and a Dragino NB IoT Bee (https://www.dragino.com/products/nb-iot/item/130-nb-iot-shield.html) to try out the BG96.

treyconley, 2020-03-03:
> I've just recently acquired a Nimbelink Skywire (https://nimbelink.com/products/4g-lte-m-global-quectel/) and a Dragino NB IoT Bee (https://www.dragino.com/products/nb-iot/item/130-nb-iot-shield.html) to try out the BG96. I'm pretty sure I'd fall into what Quectel would call "not important enough."

I have looked at both of these modules. The one thing that I need in my project is voice call capabilities as well as the IoT capabilities. I had noticed in the TinyGSM module for the BG96 that the call functions were available, but I haven't found a BG96 breakout with audio input/output for this.

SRGDamia1, 2020-03-03:
Oh, I can't help with that.  Despite being one of the contributors, I've never actually tested any of the calling components on any of the boards.  It's not something I've ever had a use for.

ThePrefixCyber, 2020-03-03:
Quectel will give you an eval board.  If you're not important enough they will likely sell you one.

piyush9999, 2022-10-07:
Hello Guys, I have established websocket connection using tinygsm and I want to send a JSON data to the server, can anyone let me know how to send the JSON message to the server using tinygsm. TIA

asi-9, 2023-11-23:
> Hi, I have tested this with a Quectel EC25 ad so far the MQTT test is working ok. I will perform further testing in the coming weeks and will most likely create a new file specifically for the EC20/21/25 modems.

Working with EC25, I am unable to connect to MQTT broker using this branch with EC20 gsm support.
Can someone please provide any example code.