vshymanskyy/TinyGSM #158: Question: need help with certificates

hubaksis, 2018-04-24:
Got a 5320 module. Trying to implement a secure server connection. A new pull request will be soon :)

Basically it works fine, I can send and read data over https. However, when I break the server's certificate (set is to invalid or expired), the module still connects to the server and sends/receives data.
How I can make the module to check that the connection is secure before the connection?

I found that there is possible to upload certificates to the module.
I opened my website, downloaded 3 certificates: CA, intermediate, and site certificate.
Then I uploaded them to the modem using AT+CCERTDOWN command.
Then I am trying to use them:
```
AT+CHTTPSSTART

OK
AT+CSSLCA=0,"ca.der" --  Set the CA used in the module, 0 - index of CA in the chain. The range is from 0 to 3. 

OK
AT+CSSLCA=1,"inter.der"

OK
AT+CSSLCERT="website.der",2 -- Set the certificate file used in the module. 2 index of CA file in the chain. The range is from 0 to 3. The <filename> certificate file has been signed using this CA file. 

OK
AT+CSSLLOADCK

ERROR
```

I think I am missing something because they use client certificates with private keys everywhere in examples. Can you tell me why? As far as I understand the module just need to check that the site's certificate is valid. 

Thanks.

vshymanskyy, 2018-04-26:
@hubaksis cool! Yes It looks like you're correct. Did you work with HTTP-specific AT commands, or at TCP/SSL/TLS level?

hubaksis, 2018-04-26:
@vshymanskyy I am using https commands to setup the connection and send/receive data.
First time I tried not to load any certificates, and this code
```
AT+CHTTPSSTART --  Acquire HTTPS protocol stack 

OK
AT+CHTTPSOPSE="server_address",443,2 -- Open HTTPS session 

OK

```

connects me to the server even if the certificate is invalid.

As far as I know usually client doesn't need its own certificate to make a connection. The client should be able to check the sum of the certificate and compare it with the sum in the code (for example, HTTPS connection over WiFi works this way for ESP8266). However, I couldn't find any commands to perform this check or to receive the certificate.

Then I tried to play with certificates. I downloaded ca, intermediate, website certificates with my browser in der format, loaded into the module and tried to use them before the connection.
(I suspect that this functionality is actually to authenticate the client, not to check the server certificate, but I couldn't find any details).

I tried to load ca and intermediate certificate using AT+CSSLCA command 
```
AT+CSSLCA=0,"ca.der"
AT+CSSLCA=1,"intermediate.der"
```

I tried to add a server certificate

```
AT+CSSLCA=0,"ca.der"    -- Set the CA used in the module 
AT+CSSLCA=1,"intermediate.der"
AT+CSSLCA=1,"server.der"
```

`AT+CSSLLOADCK` (Set the certificate file used in the module) returns `OK` in both cases, but the connection is still accepted with invalid certificate.

When I try to add website certificate as a client certificate, then `AT+CSSLLOADCK` returns error. Probably because there should be a private client certificate with a private key. 

Anyway, I couldn't find any way to make the module check the website certificate. Without this there is no point in HTTPS connection.

vshymanskyy, 2018-04-27:
I suggest you to try the plain TCP/SSL API.

hubaksis, 2018-04-27:
Oh. It was so easy to do it using built in functionality for https.
I sent a message to the manufacturer regarding this issue and I'm waiting for a reply.

Do you have any example where I can get an idea how to handle HTTPS over TCP manually?
Thank you for the idea.