vshymanskyy/TinyGSM #163: TinyGSM [MQTT] on SIM800L with NodeMCU stucks...

dnksavaliya, 2018-05-07:
Hello sir,

I am using tinyGSM library on sim800L with nodemcu at voltage 4.5v.

My MQTT code start & run stable for few minutes after that GSM SIM800L hangs up, 
i don't know code stuck or any hardware issue ?

vshymanskyy, 2018-05-09:
Closing as no AT logs, sketch is provided, etc.

closed 2018-05-09 by vshymanskyy

hubaksis, 2018-05-09:
Firstly, the modules 800L I saw needed voltage 3.4-4.4V. 4.4V - maximum. So, to be on the safe side make it 3.8V. You need to check your module specification to figure our what the voltage should be.
Check your logs, probably you'll see 'overvoltage shutdown' message.
And as Volodymyr said, provide logs and sketch if you want someone to have a look at your problem.

dnksavaliya, 2018-05-10:
Hello sir,

I have attached my code here, i am not getting any at log in my arduino because GSM got hangs up.

[Feedback_sw_low.zip](https://github.com/vshymanskyy/TinyGSM/files/1990369/Feedback_sw_low.zip)