vshymanskyy/TinyGSM #164: Support of GOOUUU TECG IOT-GA6 : reflash the firmware !

manuprendlair, 2018-05-09:
Hello,

This not really an issue but a solution for these who bought IOT-GA6 from GOOUUU TECH.
You could reflash the firmware with the AI-THINKER A6 firmware.
I just do it and confirm this is working!

please look at:

http://wiki.ai-thinker.com

you will find tool and latest A6 firmware
(use google translation if like me you couldn't read chinese!)

Have Fun!

Emmanuel

welir, 2018-05-09:
If I load firmvare send "Failed"

manuprendlair, 2018-05-09:
I just did it without any problem.
I used a ftdi ttl/rs232 converter with rx to HTX (A6) and tx to HRX(A6)

This is working with the ai thinker tool

manuprendlair, 2018-05-09:
Vous etes français?

welir, 2018-05-09:
I use same  ftdi ttl,  But issue in open port 
[image]

manuprendlair, 2018-05-09:
This is working perfectly for me.
Do you check that your are able to comunicate to your FTDI adaptater ?

vshymanskyy, 2018-05-09:
Hi @manuprendlair , thanks for your input. This is valuable, added link to this topic to the README.
The issue is closed, but feel free to continue the discussion here!

gerwinvis, 2018-06-05:
i downloaded the tool and the firmware... opened the tool, connect the oit-ga6 to my computer. My computer tells me it doesn't recognize the usb device and nothing appears in the tool.
Is there a driver i need to install first? 
ps: i tried the connection with bost the h-rxd and h-txd as the u-rxd and u-txd

nothing happens

manuprendlair, 2018-06-05:
how do you connect the A6 to your computer? you need a serial to usb
converter module.
You need first to check your module ir recognize by your computer (sometime
you need to download driver, but generaly not if you use ftdi converter
module), a new com port must be create.
Sometime the tool not working if you have different com port.
also take care to your connection tx to rx and rx to tx

Generally, problems come from the user and connections, check all two
times, three times!

For myself, I had too much problem to make the A6 working correctly with at
command (gprs connection etc.), so finally I decide to use a sim800 module,
who is a sure value!

Have fun!

manuprendlair, 2018-06-05:
> tnxs
> driver for PL2303HXD usb to uart ttl is installed and recognized bij the
> computer
> and the com port thing you explain is new for me....
>
> i think i'm gonna buy an other one... haha

this usb serial converter is best : FTDI FT232RL USB to TTL Serial
Converter Adapter Module (google this)

For the gprs, yes I suggest not to spent to much time with the A6 !

have fun

manuprendlair, 2018-06-05:
> allright, don't know what i did different but it says it passed.
> i'm a noob in this things... what software do you use to test with at
> commands?

humm If I well remember on the ai thinker wiki you could find a tool to
test at command.
but I wrote my code with the gps tiny library.
I had no success with the A6, sometine register to the network, sometime
not, sometime return some code nobody know...I just was successful sending
sms and make call two or tree times..but not for gprs connection
let's me know if you have success with it !

gerwinvis, 2018-06-05:
tnxs!!

manuprendlair, 2018-06-05:
you're welcome.
Have Fun

gerwinvis, 2018-06-05:
any tips on the best baud rate?

manuprendlair, 2018-06-05:
The baudrate is 115200 for the A6
You could change it, but when reseting the baudrate come back to 115200.
Normally you could use profile to save settings like baudrate. profile must
normally be load at startup... I never been able to make it working on A6!

Good luck and let's me know if you'll have success

janobstruggler, 2018-06-18:
Experts Please Help! I can not even connect and run the Goouuu Tech IOT-GA6 module? I am looking literature for this module but all of them are in Chineese, please help me to understand and read about this module in english links, pdfs etc.

gerwinvis, 2018-06-18:
To end my story, i stopped trying. Bought a adafruit feather fona. Sorry, its not an answer to your question but i wanted to end my story

janobstruggler, 2018-06-19:
[image]

so the gsm model which is paining me is on the picture

cbolanos79, 2018-06-19:
I tried a IOT-GA6 module with arduino mega 2650 using Serial1 instead SoftwareSerial

Using arduino for connection between modem and computer worked ok: AT commands worked right, but I got errors when tried GPRS connection sequence.

It failed every time until I found the right sequence:

Instead using

AT+CSTT=”apn”, “”, “”

Use

AT+CGDCONT=1,”IP”,”apn”

Ok, I use an apn without user and password, but it would be nice if anybody could try connection to A6 module and GPRS connection

I've no time for now to modify library and try gprs connection, but it should work.

yatan, 2018-06-27:
Worked TinyGSM with Goouuu Tech IOT-GA6.

Updated a Goouuu Tech IOT-GA6 module using a stable oficial A6 firmware using simple usb_ftdi_ttl converter, connecting **vcc_in, gnd, rst, h_rxd, h_txd pins** and work fine the upgrade.

Then used a AT_Debug for know the modem COM speed (after upgrade changed to 38400), and upgrade de IMEI with AT commands commented the user: @manuprendlair 

Used Diagnostics sketch to test, and work some GET petition to web server, the modem get stuck to recognize the response, but the petition to server are made correctly.

lucianog2000, 2018-06-29:
Hola, tengo este módulo y necesito ayuda para un proyecto ya que no estoy muy adentrado en el tema de módulos gsm y gprs. No encuentro nada que explique paso por paso como hacerlo funcionar y por eso acudo acá. Si alguien puede darme una mano estaría muy agradecido. Desde ya muchas gracias.

GerardoAllende, 2018-07-01:
The only way it worked for me was updating the firmware as @yatan posted but connected FTDI TX pin with the A6 U_TX and the RX with U_RX. This way I can receive the "OK" response from my "AT\r."

@lucianog2000 yo también estoy empezando con este módulo y perdí todo el día probando. En mi perfil está mi dirección de email.

cbolanos79, 2018-07-04:
@yatan are you using software serial? If so, try to set baudrate 9600. Software serial may not be safe for higher baud rate.

chaton78, 2018-07-25:
Firmware upgrade worked for me... but I had to connect RX to HRX and TX to HTX which is the opposite of what I am used to with Serial Connection. Note that Uart connection is standard on this module (RX->U_TX, TX->U_RX). Don't forget the Ground.

maxholgasson, 2018-07-29:
Hi, unfortunatly I also got a GOOUUU TECG IOT-GA6 instead of a AI A6

i updated now the board with the AU Thinker firmware sucessfully when sending "ATI" I got the current info now

`ATI

Ai Thinker Co.LTD
A6

V03.06.20171127R

OK`

However I still got an error when trying to send a sms:
`AT+CMGF=1.

ERROR`

Tried different AT commands and then get error 58:

`CINIT

+CME ERROR:58
AT + CREG?

+CME ERROR:58`

Vcelist, 2018-10-21:
Hi! I flashed this module (from Goouu Tech) firmware from Ai Thinker and it began to work at a speed of 115200 (previously, before the firmware, it worked fine on 9600) but it stopped registering in the gsm network when connected to Arduino (whem disconnet - register and receive sms ok). So I would like to flash it back-the original firmware from Gooouu tech-can anyone find it? I couldn't find(((

yknivag, 2018-10-28:
Is it possible to flash the firmware on these devices without a Windows PC?  It looks like the AI-Thinker tool is only available for Windows.  Is there a Linux/python solution?  Or a command line one can issue over the serial?

yknivag, 2018-10-30:
After much searching I have managed to track down both a datasheet and AT command list for the Goouuu Tech Ltd IOT-GA6-B device here http://www.lingzhilab.com/forum.php?mod=viewthread&tid=29

Unfortunately a Chinese mobile phone number is needed to register with the site to download the files so I am still no nearer getting it to work, but if anyone who has a Chinese phone number can register and download them then maybe it will be possible to get it working with this library...

chijiao, 2018-11-06:
> After much searching I have managed to track down both a datasheet and AT command list for the Goouuu Tech Ltd IOT-GA6-B device here http://www.lingzhilab.com/forum.php?mod=viewthread&tid=29 [...]

I'm chinese, i had all files about you,  
https://drive.google.com/open?id=1ikYhlcg3KJLbDsWzYDg7RmtgruROlYFk

AT command pdf file
https://drive.google.com/open?id=1keBj3Ht6U5wnm6jFlwNSaO8__CXJmFWy

andrids95, 2018-11-06:
Hai, i tried to flash my IOT-GA6 using that Ai thinker tools but no port detected... 
[image]
i used the old tools and the new one, its just same... no port detected
but in device manager my port actually detected ... 
[image]
and i can send AT command
[image]
so i think my port just working fine, but why Ai Thinker tools can't detect my port?
i used FTDI and USB to TTL, and the result is still same... Any Solution?

chaton78, 2018-11-06:
You have to reverse the connection for programming and it is not the same
pins.

andrids95, 2018-11-06:
> 
> […](#)

i've done that, just nothing happend my port still didn't detected... i've tried to reinstall the driver too, and im using windows 10 64bit.  is that normal if the port not listed in Ai Thinker tools?

chijiao, 2018-11-07:
update firmware not use rx tx，a6
have another pins, htx hrx, this two pins for update firmware，please check

chijiao, 2018-11-07:
http://wiki.ai-thinker.com/gprs_download

GPRS系列模组一般有两个串口，一个AT串口，一个下载串口。我们使用Hst_txd、Hst_rxd引脚下载程序。
使用USB-TTL接在模组的Hst_txd、Hst_rxd引脚，将串口连接到电脑上。

This module have two uart, one for mcu, one for download firmware, use usb-ttl connect Hst_txd、Hst_rxd, it's ok, i had tested

procopio, 2018-12-17:
> > […](#)
> 
> i've done that, just nothing happend my port still didn't detected... i've tried to reinstall the driver too, and im using windows 10 64bit. is that normal if the port not listed in Ai Thinker tools?

I have the same problem @andrids95!

The important thing here is we are talking about a another board its the B one. GA6-B is different from GA6. With GA6 I can reflash without problem but this GA6-B is not detected by any version of Download Tool.

ideunkr, 2019-01-04:
> Worked TinyGSM with Goouuu Tech IOT-GA6.
> 
> Updated a Goouuu Tech IOT-GA6 module using a stable oficial A6 firmware using simple usb_ftdi_ttl converter, connecting **vcc_in, gnd, rst, h_rxd, h_txd pins** and work fine the upgrade.
> 
> Upgrade tool:
> http://wiki.ai-thinker.com/_media/ai-thinker_gsm_module_download_tool_v11.00.04.7z
> Upgrade image:
> http://wiki.ai-thinker.com/_media/gprs/a6_v03.03.20161229019h03.rar
> 
> Then used a AT_Debug for know the modem COM speed (after upgrade changed to 38400), and upgrade de IMEI with AT commands commented the user: @manuprendlair
> 
> Used Diagnostics sketch to test, and work some GET petition to web server, the modem get stuck to recognize the response, but the petition to server are made correctly.

where the "rst"  Pin connected to

phuongnv9, 2019-01-09:
Hi! I flashed this module (from Goouu Tech) firmware from Ai Thinker and it's ok. But I can not display sms as text mode (AT+CMGF=1). Help me please!
[image]

xthuan, 2019-01-27:
Any update regarding the GA6-B board? The PWR pin of B board stays LOW when system boots up, it is the major difference. Anything related to this? Basically GA6-B is still a GA-6.

procopio, 2019-01-30:
> Any update regarding the GA6-B board? The PWR pin of B board stays LOW when system boots up, it is the major difference. Anything related to this? Basically GA6-B is still a GA-6.

Until now there is no way to reflash GA6-B with Ai Thinkers firmware. It comes officially from factory. Try it as is, ti me the GA6-B is working well with original firmware. 

Good luck!

procopio, 2019-01-30:
> Better check the baudrate of GA6, default is 115200 I believe.

Confirmed the default is 115200.

xthuan, 2019-01-30:
> > Any update regarding the GA6-B board? The PWR pin of B board stays LOW when system boots up, it is the major difference. Anything related to this? Basically GA6-B is still a GA-6.
> 
> Until now there is no way to reflash GA6-B with Ai Thinkers firmware. It comes officially from factory. Try it as it is, to me the GA6-B is working well with original firmware.
> 
> Good luck!

Thanks for your reply:) 
I can use AT commands to establish TCP connection, however, can not use MQTT through PubSubclient on the top of TinyGSM.

Any comments?

friistf, 2019-02-02:
Hi,
I have some (easy?) questions:
1. When I want to connect to sent AT-commands, do I connect to U_TXD/U_RXD or to H_TXD/H_RXD.
2. At which baud rate do I communicate?
3. When I send an AT-command, do I suffix the "AT" with CR - like "AT$13" 
I think these are the questions that have to be answered first.
Can anyone help?
best regards
torben

xthuan, 2019-02-03:
> Hi,
> I have some (easy?) questions:
>
> 1. When I want to connect to sent AT-commands, do I connect to U_TXD/U_RXD or to H_TXD/H_RXD.
> 2. At which baud rate do I communicate?
> 3. When I send an [...]

1. U_TXD/U_RXD
2. 115200
3. When you use Arduino IDE, type "AT" and don`t forget to select the "NL and CR" option.

-Xintao

xthuan, 2019-02-03:
> Hi xthuan,
> Thank you for your answer. You write:
> "When you use Arduino IDE, type "AT" and don`t forget to select the "NL and CR" option"
> 
> I am using Picaxe, so my communication after points 1. and 2. will look like this:
> 
> hSerOut 0, ( "AT" )
> hSerIn (1000, MAIN), 0, 4
> 
> where the latter should result in the answer "OK". Is that right?
> Since I dont use Aduino I dont have the options "NL" and "CR".
> best regards
> torben

Seems correct.

friistf, 2019-02-03:
Hi xthuan,
OK,
There are 2 lights:
One that is steady red (pwr?)
one that is blue and flashes approx. every 3 sec. 
What do they show?
The blue light was at a time flashing rather fast.
torben

HebaEbrahem, 2019-02-22:
> [image]
> 
> so the gsm model which is paining me is on the picture

Heeey , i have the same module and i'm at the very early phase of wiring it to the PC to enable communication  , have you figured out how ?

xthuan, 2019-02-23:
> Hi xthuan,
> OK,
> There are 2 lights:
> One that is steady red (pwr?)
> one that is blue and flashes approx. every 3 sec.
> What do they show?
> The blue light was at a time flashing rather fast.
> torben

The red light indicates the power, and blue light stands for mainly for connecting actions -- e.g., finding network, sending data, etc.

--xintao

xthuan, 2019-02-23:
> > [image]
> > so the gsm model which is paining me is on the picture
> 
> Heeey , i have the same module and i'm at the very early phase of wiring it to the PC to enable communication , have you figured out how ?

A good news is that, your board is the regular GA6 instead of the GA6-B, which means, you can use the aforementioned tutorial to flash the GA6 module to aithinker-A6 firmware.

--xintao

friistf, 2019-02-23:
Hi,
I have to connect hSerOut to U-TXD and hSerIn to U-RXD t make the AT commands work. Is that normal?.
torben

xthuan, 2019-02-24:
> Hi,
> I have to connect hSerOut to U-TXD and hSerIn to U-RXD t make the AT commands work. Is that normal?.
> torben

Normally should be MCU_Tx - GSM_URx   & MCU_Rx - GSM_UTx.

HebaEbrahem, 2019-02-25:
Helloo, did anyone happen to be using python to send AT-commands to the module through the usb-ttl converter  ?  if yes what library ? (windows OS) .thanks in advance

HebaEbrahem, 2019-02-25:
Also what IDE that i can use (if i'm not using arduino ) ,to configure the module with AT-commands ?
through the usb-ttl .

xleray, 2019-03-17:
Thanks all you Guys for this great info and Help

belalmc, 2019-04-14:
@xthuan GA6-B chip is not the same as GA6 Chip, 
GA6:      RDA88X9CL  
GA6-B :  RDA8955L

nick4o, 2019-05-20:
> Hai, i tried to flash my IOT-GA6 using that Ai thinker tools but no port detected...
> [image]
> i used the old tools and the new one, its just same... no port detected
> but in device manager my port [...]

Hi everyone I'm new with this. Is there solution for this issue? my one is IOT-GA6-B and Ai-thinker can't find the port to communicate with the board.
I'm using the following pins:
HTX
HRX
GND
VCC
Tanks in advance.

xthuan, 2019-05-20:
> Hi everyone I'm new with this. Is there solution for this issue? my one is IOT-GA6-B and Ai-thinker can't find the port to communicate with the board.
> I'm using the following pins:
> HTX
> HRX
> GND
> VCC [...]

Unfortunately still no solution on that. The GA6-B is quite problematic.

andrids95, 2019-05-21:
> Unfortunately still no solution on that. The GA6-B is quite problematic.

you may need to consider using another module like sim800 or sim900.  I've done with GA6-B with no solution at all and change to another gsm module

nick4o, 2019-05-21:
Thanks very much

I’m giving up as well.

Regards
Nik

xleray, 2019-06-18:
I managed to communicate with mine
it might sound weird, but it worked for me;
use; 
Vcc_in -----------------5v
U_RXD --------------- RX
U_TXD---------------- TX
GND--------------------GND

nick4o, 2019-06-18:
Thanks I’ll try

petrilloa, 2019-07-09:
Hi everyone.
Has anyone succeeded with GA6-B? I need to do a project with a GSM chip at 3.3v, already discarding SIM800L and similar boards.
I was testing A6 from Ai-Thinker and it works correctly, but I need the MINI version. My supplier is only able to get this version of GOOUUU TECH and I am afraid that it will not work.

[14 more replies]

commit ffe4611f (2018-05-09): Update README.md

Fix #164