vshymanskyy/TinyGSM #166: [SIM800] SSL client does not read the data, sent just before socket is closed

vshymanskyy, 2018-05-09:


vshymanskyy, 2018-05-09:
This restricts ability to download files using HTTPS.
But, there is a simple workaround:
1. Tell the server not to disconnect by providing `Connection: keep-alive` request header.
2. Ensure your server sends `Content-Length` header in response.
3. Safely disconnect client when the full body is received.

Profit.

ianarchbell, 2018-07-26:
Hi, first all thanks for the library - works great non-SSL and SLL GET works fine too. However, this workaround doesn't seem to work for the general POST case where the output is chunked. It looks like the first chunk gets read but not the following. Keep-alive makes no difference. You can try any POST to scriptr.io - they all fail as far as I have determined. Output below shows data available but not fully read - in this example the first chunk is 63 bytes - it does read read that but doesn't read the second chunk. By the time http.responseStatusCode() is called the modem is closed and so times out. If you have any another example that shows a POST using SSL with chunked data output I'd be happy to try it. Thanks, Ian.

DATA ACCEPT:1,90
End of post
waitResponse

+CIPRXGET: 1,1
modemGetAvailable
AT+CIPRXGET=4,1
waitResponse

+CIPRXGET: 4,1,328
waitResponse

OK
modemRead
AT+CIPRXGET=2,1,63
waitResponse

1, CLOSED

vshymanskyy, 2018-07-27:
@ianarchbell unfortunately it looks like a SIM800 bug. I don't know if there is any other workaround, but if someone finds something - we can add it to the library.