vshymanskyy/TinyGSM #169: Receive SMS function for Neoway N10

Santosh1971, 2018-05-23:
I am using Neoway GSM modem N10 by modifying SIM800 driver. I will be greatful if you can include N10 also in the driver. If it takes more time then at least help me for the function for receiving SMS.I am attaching the AT commands document of N10

[Neoway_N10_AT_Command_Manual_V1.5(1).pdf](https://github.com/vshymanskyy/TinyGSM/files/2029336/Neoway_N10_AT_Command_Manual_V1.5.1.pdf)

vshymanskyy, 2018-07-01:
Neoway N10 is not yet supported.

vshymanskyy, 2018-07-02:
> here is the attachment and the AT commands of N10.
>
> On Mon, Jul 2, 2018 at 12:54 PM Santosh Kr. jha <jha.santosh.kr@gmail.com>

So.. how does it help with having hardware at hands.? ;)

Santosh1971, 2018-07-02:
Yaa, I can understand the difficulty in doing this without Hardware.

Anyway , I will try and send you the code ,
I will be happy to ship my hardware to you if you prefer.
regards,'
Santosh

vshymanskyy, 2018-07-03:
@Santosh1971 can you send me the link to your board/vendor? It should be easy to get a sample from them directly.

Santosh1971, 2018-07-03:
http://www.iot-store.uk/index.php/neoway-n10-evaluation-kit.html
https://www.dhgate.com/product/n10-gprs-module-gsm-mqtt-wireless-data-transmission/410017149.html

Santosh1971, 2018-09-09:
Hello Volodymyr,
Could you get chance to work on N10?
Regards,
Santosh