vshymanskyy/TinyGSM #171: Problems on Esp32 and SIM900 modem

euquiq, 2018-05-30:
TinyGSM version: LATEST
Hardware: ESP32 ARDUCAM, SIM900 SEEEDstudio v2.0 "shield" modem.

I am powering the modem with 5V directly from external 7805.
I am connecting ESP32 through hardwareSerial (on pins 17,16)

I am controlling the POWER button through shield's PIN9, 

When using TinyGSM in this way, only the first AT command gets answered thru TinyGSM. Then Sim900L stops responding through tinyGSM.

I.E.: When this code is run, the first thing is to check if SIM900 modem is powered up or not.

This happens inside gsm_power() and when the first  modem.testAT() is run, it returns, accordingly a false value if timeout is reached or true, if the AT command is answered.

So, accordingly, my code keeps running, powering on (or stating that the modem is already powered up).

AND THEN, tinyGSM stops contact with the SIM900 modem!!

If the SIM900L modem needed a POWER UP, I can visually verify it does power up, but the second modem.testAT() keeps returning FALSE (because the AT command tinygsm internally issues to test modem availability TIMES OUT.

If the SIM900L modem was already POWERED UP, then the next code (in gsm_init()) wich queries the MODEM info (with ATI command) returns an empty string (again, tinygsm is uncapable of connecting with it.

Now for the weird thing:

I program myself a similar code, performing these same basic steps, inside my .ino file, and everything goes OK. 

So this SIM900 / Hardware Serial connection "going south" only happens on tinyGSM realms.

Any idea anyone ? I am going berzerk with this situation, as I really would love to get tinygsm functionality to work.

Regards and thanks, 

Enrique

```
#define TINY_GSM_MODEM_SIM900
#define SerialAT Serial1
#define pin_gsm_pw 13
#include <TinyGsmClient.h>

HardwareSerial Serial1(2); 
TinyGsm modem(SerialAT);
TinyGsmClient client(modem);

boolean gsm_modem = false;

void setup()
{
	Serial.begin(115200);
	pinMode(pin_gsm_pw, OUTPUT); //So I can control POWER UP / DOWN
	SerialAT.begin(115200, SERIAL_8N1, pin_gsm_tx, pin_gsm_rx,false); //GSM MODEM SIM900L
	delay(3000);
        gsm_init();
}

void loop() {
 //Etc etc ...
}

void gsm_init() {
	gsm_power();
	Serial.println("Modem: " + modem.getModemInfo());
	//modem.simUnlock("1757");
	Serial.print("Waiting for network...");
	if (!modem.waitForNetwork()) {
		Serial.println(" fail");
		while (true);
	}
	Serial.println(" OK");

	Serial.print("Connecting to ");
	Serial.print(apn);
	if (!modem.gprsConnect(apn, user, pass)) {
		Serial.println(" fail");
		while (true);
	}
	Serial.println(" OK");
}

void gsm_power() {
	Serial.println("GSM INIT");
	gsm_modem = modem.testAT();
	if (!gsm_modem) { //IT'S OFF ...
		Serial.println("POWERING GSM ON!");
		digitalWrite(pin_gsm_pw, LOW);
		delay(1000);
		digitalWrite(pin_gsm_pw, HIGH);
		delay(2000);
		digitalWrite(pin_gsm_pw, LOW);
		delay(4000);
		gsm_modem = modem.testAT(); //CHECK POWER NOW!
		Serial.println("GSM_MODEM: " + (String)gsm_modem);
	}
	else {
		Serial.println("GSM ALREADY ON");
	}
}

```

euquiq, 2018-05-31:
UPDATE:

It seems to be related with the usage of String class and FreeRTOS inside esp32 Arduino:

https://github.com/vshymanskyy/TinyGSM/issues/86

vshymanskyy, 2018-06-14:
@euquiq I don't have SIM900, so can't investigate this. I suggest you to investigate it further and provide as much information as you can, what exactly goes wrong from TinyGSM perspective. Then we could think about a solution. Also be sure to use the latest ESP23 core, etc.

vshymanskyy, 2018-07-01:
Closed due to lack of interest

closed 2018-07-01 by vshymanskyy