vshymanskyy/TinyGSM #174: Question: Whats needed for compatibility with sim5360

ErikssonEivind, 2018-06-08:
TinyGSM version: latest
Hardware: esp32, SIM5360
*Power is stable. Using a external PSU

Hi! 

I'm trying to configure tinyGsm to be able to use it with a esp32 and sim5360. 

I used the AT_Debug sketch to verify that the sim5360 is working. Receiving "OK".
My end goal is to use tinyGSM with the pubsubclient library. I've managed to do some basic MQTT sending and receiving with the SIMCOM MQTT AT commands. However I would love to be able to use the tinyGsm+pubsubclient library instead. My questions is: 

1) Is it possible to modify the current tinyGsm for use with sim5360?
2) Is it known what is needed to use tinyGsm with sim5360?

I would gladly try to help out with making tinyGsm compatible with sim5360 if I just had an idea of where to start :)

I

closed 2018-06-14 by vshymanskyy

NatroNx, 2018-07-08:
any news on this? Would be awsome to test

laurentvm, 2018-07-26:
Hi,

Freematics has to source files using both SIM800 and SIM5360, isn't enough to compare both implementation to implement a TinyGSM for SIM5360

https://github.com/stanleyhuangyc/Freematics/blob/master/libraries/FreematicsPlus/FreematicsNetwork.cpp