vshymanskyy/TinyGSM #182: [QUESTION] secure MQTT broker

marabesi, 2018-07-22:
I've been using this library with the mosquitto broker without problems (via 1883 port). Now I'd like to secure my server and the data that I am using, the problem is that it seems I can't connect in a broker using ssl (port 8883).

Is there a way to connect to a server using ssl?

I am not sure if this is a question for tinyGSM project or pubsub library.

Thank you

wbmrcb, 2018-07-23:
Try `TinyGsmClientSecure client(modem);` as in HttpsClient example

marabesi, 2018-07-27:
Hello @wbmrcb it seems to be working the solution you gave me. 

I believe this could be a reference for future questions about ssl.

Thank you

closed 2018-07-27 by marabesi

Eiyvor, 2020-07-14:
This is not working for me though.

SRGDamia1, 2020-07-20:
@Eiyvor - if you want help, you need to provide more detail about what isn't working.

Eiyvor, 2020-10-13:
@SRGDamia1  I first uploaded the SSL to the sim800 module successfully, then in another sketch I used the TinyGSMClientSecure class (hopping that the class would use the certificate I uploaded, while connecting to the broker) and connected to the secure port 8883. I used the example sketch provided by you to upload the certificate.

SRGDamia1, 2020-10-27:
You're definitely able to make a secure connection and send and receive data over it?  (ie, you've successfully run the HTTP)  If you cannot do that, you're probably not going to be able to get a secure MQTT connection.  The SIM800 is pretty iffy with SSL, a lot of people have trouble with it.

But if you want more help, you need to share more detail about what's not working.