vshymanskyy/TinyGSM #183: Error when compile in the STM32 platform

daviidmart, 2018-07-23:
[image]

daviidmart, 2018-07-24:
PLATFORM: ST STM32 > STM32F103T8 (20k RAM. 64k Flash)
SYSTEM: STM32F103T8T6 72MHz 64KB RAM (20KB Flash)
DEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TinyGSM> 0.3.5 (C:\Users\david\Documents\PlatformIO\Projects\IVO FIRMWARE\.piolibdeps\TinyGSM_ID1287)
|-- <Blynk> 0.5.3 (C:\Users\david\Documents\PlatformIO\Projects\IVO FIRMWARE\.piolibdeps\Blynk_ID415)
|   |-- <TinyGSM> 0.3.5 (C:\Users\david\Documents\PlatformIO\Projects\IVO FIRMWARE\.piolibdeps\TinyGSM_ID1287)
arm-none-eabi-g++ -o .pioenvs\development\firmware.elf -T jtag_t8.ld -Wl,-u,_printf_float -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m3 .pioenvs\development\src\main.cpp.o -LC:\Users\david\.platformio\platforms\ststm32\ldscripts -L.pioenvs\development -LC:\Users\david\.platformio\packages\framework-arduinoststm32\STM32F1\variants\generic_stm32f103t\ld -Wl,--start-group .pioenvs\development\libbbe\libTinyGSM_ID1287.a .pioenvs\development\lib152\libBlynk_ID415.a .pioenvs\development\libFrameworkArduinoVariant.a .pioenvs\development\libFrameworkArduino.a -lc -lgcc -lm -Wl,--end-group
c:/users/david/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7-m\libnosys.a(sbrk.o): In function `_sbrk':
sbrk.c:(.text._sbrk+0x18): undefined reference to `end'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\development\firmware.elf] Error 1

vshymanskyy, 2018-07-27:
You should report this to platformio

closed 2018-07-27 by vshymanskyy