vshymanskyy/TinyGSM #186: ESP32 Unhandled exception in TinyGSM

RoboDisko, 2018-07-28:
TinyGSM version: Latest (0.3.5)
Hardware: Heltec_WiFi_LoRa_32 with A6 modem (Ai Thinker Co.LTD A6  V03.03.20161229019H03)

I'm using a MQTT client (PubSubClient) that is calling something in TinyGSM that is causing an exception. The issue is intermittent and random. It usually happens during setup when setting up the MQTT connection, but sometimes during the main loop. If i let it happen enough times, it usually goes away.

This is what I get from serial: (with AT dump)
https://gist.github.com/RoboDisko/b0c90b2fbb73d514bc03d8c7df30dce1

This is the crash stack trace results:
https://gist.github.com/RoboDisko/440bfe082c0ae0d960939fd28575f530

Any ideas how to make that not happen?

SRGDamia1, 2020-04-15:
I think I just fixed this.

closed 2020-04-15 by SRGDamia1

rpolania, 2021-01-21:
Hi SRGDamia1. I have the same issue. Could you please tell me how do you fixed it?

Thanks.

SRGDamia1, 2021-01-22:
@rpolania Are you using the latest version?  It should have been already fixed.  If you're using an ESP32, can you upload the stack dump?

commit fb3f8583 (2020-04-15): Some more null pointer checks, fix #186