vshymanskyy/TinyGSM #187: Support for watchdog reset

micw, 2018-07-29:
TinyGSM version: latest
Hardware: SIM800

### Scenario, steps to reproduce
I have a watchdog running that resets during gprsConnect. I run on a STM32 board where max watchdog timeout is ~25s.

### Expected result
I want to have the watchdog reset if gprsConnect is working properly.

### Actual result
gprsConnect takes often too long, so the watchdog resets the board.

### AT command log
---

Proposal: add an option to run a custom callback within waitResponse, so that user-defined code (i.e. feed the watchdog) can be run.

vshymanskyy, 2018-08-24:
you should be able to redefine Arduino `yield()` function. or just replace the delays/loops with your own version that will ping the watchdog. Or use RTOS to put TinyGSM processing on a separate thread.

cimba007, 2018-09-22:
This should give you an idea on how to do this:
```
unsigned long last_wdt_reset = 0;
#include <avr/wdt.h>
#define TINY_GSM_YIELD() {  delay(0); wdt_reset(); if(millis()-last_wdt_reset > 2000) { SerialMon.print("\r\n[WDT-WARNING: "); SerialMon.print(millis()-last_wdt_reset); SerialMon.print("]"); } last_wdt_reset = millis(); }
```

micw, 2018-09-22:
@cimba007 Thanks for advice, I'll keep this in mind. Since I had multiple points in my project where the stm32 module stopped and I had no time to debug it, I decided to add a hardware watchdog (I used an old esp8266 module where wifi was broken) - and programmed this to much longer rest times (>1 minute) which also solved my problem with watchdog not reset during gsm transfer.

closed 2018-09-22 by vshymanskyy