vshymanskyy/TinyGSM #188: Factory reset on every reboot

rajibgupta, 2018-08-01:
Latest TunyGSM Ver
Hardware SIM800

I tried using the latest addition of getting the time from SIM800 RTC. However it kept giving me the wrong time. Something like 04/01/01...

I noticed in the library ,inside TINYGSMCLIENTSIM800.H file the restart() function

bool restart() {
    if (!testAT()) {
      return false;
    }
    //Enable Local Time Stamp for getting network time
    // TODO: Find a better place for this
    sendAT(GF("+CLTS=1"));
    if (waitResponse(10000L) != 1) {
      return false;
    }
    sendAT(GF("&W"));
    waitResponse();
    sendAT(GF("+CFUN=0"));
    if (waitResponse(10000L) != 1) {
      return false;
    }
    sendAT(GF("+CFUN=1,1"));
    if (waitResponse(10000L) != 1) {
      return false;
    }
    delay(3000);
    return init();
  }
 
It first sets SIM800 to get local time . sendAT(GF("+CLTS=1"))

At the end it calls init() function.

bool init() {
    if (!testAT()) {
      return false;
    }
    sendAT(GF("&FZ"));  // Factory + Reset
    waitResponse();
    sendAT(GF("E0"));   // Echo Off
    if (waitResponse() != 1) {
      return false;
    }
    getSimStatus();
    return true;
  }

This functions executes factory reset. I think that's just deleting all the changes we did for getting the local timestamp.

I tried commenting these two lines inside the library.

sendAT(GF("&FZ"));  // Factory + Reset
    waitResponse();
Now it gives correct time. 

So is it really a bug? Or I am just doing everything wrong?

TobbeG, 2018-08-01:
I had same issue, never got getting time from SIM800L to work... 

so I skipped it and called a webpage to get timestamp instead.

Would rather get this work, though.

rajibgupta, 2018-08-01:
As I have mentioned I got it working by skipping a few lines inside init() function. But not sure what is the correct way.

SRGDamia1, 2020-04-15:
This has been addressed, I believe.

closed 2020-04-15 by SRGDamia1

pinkfloydnow, 2022-01-10:
after commenting the two lines inside the library sendAT(GF("&FZ"));  // Factory + Reset
    waitResponse();
time is given correctly thank's to  rajibgupta .