vshymanskyy/TinyGSM #192: USSD alphabet

marmotton, 2018-08-10:
Hello,
I tried the sendUSSD function, but I get an empty string. I could reproduce this using AT commands:
```
at+cscs="HEX"
at+cusd=1,"*147#"
Response: +CUSD: 2

at+cscs="IRA"
at+cusd=1,"*147#"
Response: +CUSD: 0, "Votre cr�dit  s'�l�ve � CHF 3.75.", 2
```

The character set is hard coded inside the sendUSSD function [TinyGsmClientSIM800.h](https://github.com/vshymanskyy/TinyGSM/blob/74a4bf5d4a6016597447ed5fc629b017dfe2e8f2/src/TinyGsmClientSIM800.h#L658).

I guess there is a reason behind using HEX here, but in my case I need to change it to IRA in order to get the response string.

TinyGSM version 0.3.5 (downloaded via PlatformIO)
SIM800L (SIM800 R14.18), ESP32, Arduino framework
Carrier: Talktalk (Switzerland, Sunrise network)

marmotton, 2018-08-10:
I did a quick check to make sure that the way I was (manually) sending the AT command is not faulty.
- git clone TinyGSM
- use the AllFunctions example
- 1st try without modifications
- 2nd try, changing line 658 of TinyGsmClientSIM800.h to `sendAT(GF("+CSCS=\"IRA\""));`

Result of 1st try:
```
[3029] Initializing modem...
[12917] Modem: SIM800 R14.18
[12918] Waiting for network...
[19929] Network connected
[19929] Connecting to internet
[28652] GPRS status: connected
[28709] CCID: ******
[28751] IMEI: ******
[28800] Operator: sunrise
[28845] Local IP: 10.37.12.242
[28882] Signal quality: 23
[28925] Battery lavel: 100
[30969] Battery voltage: 4.22
[32252] GSM location: 0,******,******,2018/08/10,18:44:02
[32310] GSM Time: 01:26:58+00
[32369] GSM Date: 04/01/01
[38182] Balance (USSD): 
[44990] Phone number (USSD): 
[45471] GPRS disconnected
[45979] Poweroff.
```

Result of 2nd try:
```
[3029] Initializing modem...
[12912] Modem: SIM800 R14.18
[12912] Waiting for network...
[19342] Network connected
[19342] Connecting to internet
[26088] GPRS status: connected
[26147] CCID: ******
[26190] IMEI: ******
[26240] Operator: sunrise
[26285] Local IP: 10.150.121.22
[26321] Signal quality: 22
[26364] Battery lavel: 100
[28407] Battery voltage: 4.24
[29684] GSM location: 0,******,******,2018/08/10,18:53:35
[29742] GSM Time: 01:36:31+00
[29800] GSM Date: 04/01/01
[32959] Balance (USSD): Votre cr�dit  s'�l�ve � CHF 3.75.
[35987] Phone number (USSD): MSISDN = 4176*******
[37379] GPRS disconnected
[37883] Poweroff.
```

(I replaced some numbers with *** in the results, it was written correctly)

vshymanskyy, 2019-02-06:
Cool. Feel free to use the encoding that works for you!

closed 2019-02-06 by vshymanskyy