vshymanskyy/TinyGSM #193: Flow control (XON/XOFF)

anatom74, 2018-08-14:
For transmitting large data (text based) it would be nice to have some flow control (xon/xoff). I thought that I could do it without, using low baudrate, but it's also GPRS connection quality which influences on GSM module buffer overrun.
Anyone already done it?

vshymanskyy, 2018-08-24:
It should be just a configuration of `Serial`. Isn't it?

closed 2018-08-30 by vshymanskyy

arihantdaga, 2018-12-26:
@anatom74 Did you figure out how to do that ? Strangely i am running into same issue these days, something is happening tiny gsm library commands are recieving response from sim800 module very late. I was using stm32 bluepill with sim800 Module, figured out, this is probably be due to flow control. I checked with stm32duino arduino core about this, they mentioned flow control is not implemented. Its been long and i was wondering if you figured out any solution for this... ?