vshymanskyy/TinyGSM #197: [SIM800] Looping, Freeze, Remove recursive call to waitResponse() that removes unprocessed data

lafbrazil, 2018-08-22:
Thanks for explanation how to use or register an issue inside platform. My apologies and will be followed the instructions.

TinyGSM version: v.0.3.5
Modem: SIM808 R14.18
ATmega2560
#define TINY_GSM_MODEM_SIM808
 // SIM808
 SerialAT.begin(4800);  

Previous issues:
#69 - [SIM800] Looping with ERROR AT+CIPRXGET=2,1,63
#65 - Remove recursive call to waitResponse() that removes unprocessed data…
#62 - [SIM868] library doesn't receive "1, CLOSED" message from modem
#21 - [SIM900/SIM800] ERROR LOOP after a short time
#4  - [SIM900/SIM800] Freeze after about 15 Minutes

Basically after 12 hours working fine, I receives the message Connection already open and in sequence errors to connect. The PCB is freezes and is required an manual reset to work again.

I found that the same issues (in less or bigger time) already reported and apply the same recommendation used at #65, but issue remain the same. So, I would like any idea or workaround to solve it.

[06:35:31.439] Connected
[06:35:31.439] ---Incoming Data start ... Connection 1
[06:35:31.482] Connection already open >>> ******************************************* ATTENTION
[06:35:31.517] Connected
[06:35:31.517] AT+CIPSEND=1,3
[06:35:31.562]
[06:35:31.562] ERROR
[06:35:31.562] AT+CIPSEND=1,1
[06:35:31.562]
[06:35:31.562] ERROR
[06:35:31.578] AT+CIPSEND=1,36
[06:35:31.598]
[06:35:31.626] ERROR
[06:35:31.648] AT+CIPSEND=1,9
[06:35:31.669]
[06:35:31.690] ERROR
[06:35:31.722] AT+CIPSEND=1,2
[06:35:31.722]
[06:35:31.754] ERROR
[06:35:31.778] 01B26A/1.49/13.00/14.48 HTTP/1.1
[06:35:31.819] AT+CIPSEND=1,6
[06:35:31.840]
[06:35:31.840] ERROR
[06:35:31.858] AT+CIPSEND=1,17
[06:35:31.882]
[06:35:31.882] ERROR
[06:35:31.898] AT+CIPSEND=1,1
[06:35:31.918]
[06:35:31.946] ERROR
[06:35:31.969] AT+CIPSEND=1,3
[06:35:31.988]
[06:35:32.011] ERROR
[06:35:32.028] AT+CIPSEND=1,2
[06:35:32.045]
[06:35:32.091] ERROR
[06:35:32.091] AT+CIPSEND=1,10
[06:35:32.091]
[06:35:32.123] ERROR
[06:35:32.138] AT+CIPSEND=1,2
[06:35:32.158]
[06:35:32.187] ERROR
[06:35:32.208] AT+CIPSEND=1,13
[06:35:32.228]
[06:35:32.251] ERROR
[06:35:32.269] AT+CIPSEND=1,2
[06:35:32.288]
[06:35:32.315] ERROR
[06:35:32.427] AT+CIPSEND=1,2
[06:35:32.449]
[06:35:32.475] ERROR
[... 327 lines cut ...]
[06:35:31.898] AT+CIPSEND=1,1
[06:35:31.918]
[06:35:31.946] ERROR
[06:35:31.969] AT+CIPSEND=1,3
[06:35:31.988]
[06:35:32.011] ERROR
[06:35:32.028] AT+CIPSEND=1,2
[06:35:32.045]
[06:35:32.091] ERROR
[06:35:32.091] AT+CIPSEND=1,10
[06:35:32.091]
[06:35:32.123] ERROR
[06:35:32.138] AT+CIPSEND=1,2
[06:35:32.158]
[06:35:32.187] ERROR
[06:35:32.208] AT+CIPSEND=1,13
[06:35:32.228]
[06:35:32.251] ERROR
[06:35:32.269] AT+CIPSEND=1,2
[06:35:32.288]
[06:35:32.315] ERROR
[06:35:32.427] AT+CIPSEND=1,2
[06:35:32.449]
[06:35:32.475] ERROR
[06:35:32.498]

lafbrazil, 2018-08-31:
TinyGSM: version: v.0.3.5
Modem: SIM808 R14.18
ATmega2560
#define TINY_GSM_MODEM_SIM808
// SIM808
SerialAT.begin(4800);

Scenario, steps to reproduce
#69 - [SIM800] Looping with ERROR AT+CIPRXGET=2,1,63
#65 - Remove recursive call to waitResponse() that removes unprocessed data…
#62 - [SIM868] library doesn't receive "1, CLOSED" message from modem
#21 - [SIM900/SIM800] ERROR LOOP after a short time
#4 - [SIM900/SIM800] Freeze after about 15 Minutes

Expected result
That the data can be send without Freeze after about 15 Minutes or more.

Actual result
Basically after 12 hours working fine, I receives the message Connection already open and in sequence errors to connect. The PCB is freezes and is required an manual reset to work again.

AT command log

[06:35:31.439] Connected
[06:35:31.439] ---Incoming Data start ... Connection 1
[06:35:31.482] Connection already open >>> ******************************************* ATTENTION
[06:35:31.517] Connected
[06:35:31.517] AT+CIPSEND=1,3
[06:35:31.562]
[06:35:31.562] ERROR
[06:35:31.562] AT+CIPSEND=1,1
[06:35:31.562]
[06:35:31.562] ERROR
[06:35:31.578] AT+CIPSEND=1,36
[06:35:31.598]
[06:35:31.626] ERROR
[06:35:31.648] AT+CIPSEND=1,9
[06:35:31.669]
[06:35:31.690] ERROR
[06:35:31.722] AT+CIPSEND=1,2
[06:35:31.722]
[06:35:31.754] ERROR
[06:35:31.778] 01B26A/1.49/13.00/14.48 HTTP/1.1
[06:35:31.819] AT+CIPSEND=1,6
[06:35:31.840]
[06:35:31.840] ERROR
[06:35:31.858] AT+CIPSEND=1,17
[06:35:31.882]
[06:35:31.882] ERROR
[06:35:31.898] AT+CIPSEND=1,1
[06:35:31.918]
[06:35:31.946] ERROR
[06:35:31.969] AT+CIPSEND=1,3
[06:35:31.988]
[06:35:32.011] ERROR
[06:35:32.028] AT+CIPSEND=1,2
[06:35:32.045]
[06:35:32.091] ERROR
[06:35:32.091] AT+CIPSEND=1,10
[06:35:32.091]
[06:35:32.123] ERROR
[06:35:32.138] AT+CIPSEND=1,2
[06:35:32.158]
[06:35:32.187] ERROR
[06:35:32.208] AT+CIPSEND=1,13
[06:35:32.228]
[06:35:32.251] ERROR
[06:35:32.269] AT+CIPSEND=1,2
[06:35:32.288]
[06:35:32.315] ERROR
[06:35:32.427] AT+CIPSEND=1,2
[... 329 lines cut ...]
[06:35:31.898] AT+CIPSEND=1,1
[06:35:31.918]
[06:35:31.946] ERROR
[06:35:31.969] AT+CIPSEND=1,3
[06:35:31.988]
[06:35:32.011] ERROR
[06:35:32.028] AT+CIPSEND=1,2
[06:35:32.045]
[06:35:32.091] ERROR
[06:35:32.091] AT+CIPSEND=1,10
[06:35:32.091]
[06:35:32.123] ERROR
[06:35:32.138] AT+CIPSEND=1,2
[06:35:32.158]
[06:35:32.187] ERROR
[06:35:32.208] AT+CIPSEND=1,13
[06:35:32.228]
[06:35:32.251] ERROR
[06:35:32.269] AT+CIPSEND=1,2
[06:35:32.288]
[06:35:32.315] ERROR
[06:35:32.427] AT+CIPSEND=1,2
[06:35:32.449]
[06:35:32.475] ERROR
[06:35:32.498]

SRGDamia1, 2020-04-15:
I suspect this has been fixed.  Please re-open it if you're still having problems.

closed 2020-04-15 by SRGDamia1