vshymanskyy/TinyGSM #203: Cannot use TinyGsm within another class

siredmar, 2018-09-11:
TinyGSM version: 0.3.5
Hardware: Arduino nano v3

### Scenario, steps to reproduce
I try to use TinyGSM within another class. I have narrowed it down to a minimalistic example:
https://gist.github.com/siredmar/d51ed0da5b24e5d87ba7f5528a88a74f
I create a pointer of the TinyGSM object and pass it to another class to work with it.
In setup() i connect to my SIM800L module and finally send a SMS either directly or from another object with the modem passed as pointer - see codeswitch at the end.

### Expected result
I would expect that the TinyGSM object can be accessed from any library that got its pointer passed.

### Actual result
I cannot send the SMS from the library - please see gist from above

### AT command log

...
[18332] Waiting for network...
AT+CREG?
+CPIN: READY
Call Ready
SMS Ready
+CREG: 0,2
OK
AT+CREG?
+CREG: 0,2
OK
AT+CREG?
+CREG: 0,2
OK
AT+CREG?
+CREG: 0,2
OK
AT+CREG?
+CREG: 0,2
OK
AT+CREG?
+CREG: 0,1
OK
AT+CREG?
+CREG: 0,1
OK
[19659] Network connected
AT+CMGF=1
OK
AT+CSCS="GSM"
OK
AT+CMGS="Hallo Test"
ERROR
[19681] SMS: fail

SRGDamia1, 2018-09-11:
I have had similar problems, though I've never quite pinned down the cause.  I've been suspicious that it could be something odd happening in the linker when the code compiles, though that suspicious comes as much from my dislike of everything being built into a header file as from any real evidence that it's the problem.

vshymanskyy, 2018-10-15:
@siredmar   everything in the header is absolutely OK in C++, especially if it is used by a single CPP file. The whole STL/boost is (mostly) a bunch of headers.
I don't think anyone here is willing to investigate general C++ programming issues, so closing this. I also think "the minimal sketch" could be even more simple, if the issue was in the library. For example, all the `modem->*` commands work, which means the library is working by pointer.

closed 2018-10-15 by vshymanskyy

siredmar, 2018-10-19:
Just for the records. I never claimed that the header only implementation would be a problem.

SRGDamia1, 2018-10-19:
The header-only problem implementation of TinyGSM just messes me up because I'm not a very good programmer.  But, anyway, TinyGSM definitely does work inside another class.  I've built a whole library wrapping it.

The problem with your code is that your "sendSms" function in your "gms" class reversed the order of the number at the message itself.  So in line 86 you are trying to send message containing the word "NUMBER" to the phone number "hallo test."  Of course, that fails.  So the SIM800 responds with "ERROR."