vshymanskyy/TinyGSM #208: error bool get_GPS

ilbonch, 2018-10-10:
part code library:

```
bool getGPS(float *lat, float *lon, float *speed=0, int *alt=0, int *vsat=0, int *usat=0) {
    //String buffer = "";
    char chr_buffer[12];
    bool fix = false;

    sendAT(GF("+CGNSINF"));
    if (waitResponse(GF(GSM_NL "+CGNSINF:")) != 1) {
      return false;
    }

    stream.readStringUntil(','); // mode
    if ( stream.readStringUntil(',').toInt() == 1 ) fix = true;
    stream.readStringUntil(','); //utctime
    *lat =  stream.readStringUntil(',').toFloat(); //lat
    *lon =  stream.readStringUntil(',').toFloat(); //lon
    if (alt != NULL) *alt =  stream.readStringUntil(',').toFloat(); //lon
    if (speed != NULL) *speed = stream.readStringUntil(',').toFloat(); //speed
    stream.readStringUntil(',');
    stream.readStringUntil(',');
    stream.readStringUntil(',');
    stream.readStringUntil(',');
    stream.readStringUntil(',');
    stream.readStringUntil(',');
    stream.readStringUntil(',');
    if (vsat != NULL) *vsat = stream.readStringUntil(',').toInt(); //viewed satelites
    if (usat != NULL) *usat = stream.readStringUntil(',').toInt(); //used satelites
    stream.readStringUntil('\n');

    waitResponse();

    return fix;
  }
```

in the bool getGPS, *speed is before *alt
in the stream, *alt is before *speed

I'm doing it wrong?

ilbonch, 2018-10-10:
And:

bool getGPS(........*lat, ........*lon, ........ *speed=0, **int** *alt=0, ......... *vsat=0, ........*usat=0) {

ilbonch, 2018-10-10:
And:

`bool getGPS(float *lat, float *lon, float *speed=0, int *alt=0, int *vsat=0, int *usat=0) {`

alt is in INT

`if (alt != NULL) *alt =  stream.readStringUntil(',').toFloat(); //lon`

alt is in FLOAT

brvdg, 2018-10-11:
Hello,

I don't understand this issue. 
What doesn't work?

ilbonch, 2018-10-15:
problem A:  if i call `&gps_speed`, i get altitude!

problem B: if i call

 `bool gps_fixstatus = modem.getGPS(&gps_latitude, &gps_longitude, &gps_speed, &gps_alt);`, 

i get error because an int value is called, when in the function 

`if (alt! = NULL) * alt = stream.readStringUntil (','). toFloat (`), 

is called a float.

I hope I managed to make you understand       :) :) :)

SRGDamia1, 2020-04-15:
This has been re-written and hopefully fixed.

closed 2020-04-15 by SRGDamia1