vshymanskyy/TinyGSM #209: [Hardware] ESP32 + Sim800L - getting garbage on serial when making call

toomasz, 2018-10-12:
_This issue is not specific to this library but hopefully somebody here can help me._

My setup is: 
GSM tx <-> ESP32 14 pin
GSM rx <-> ESP32 12 pin

`Serial2.begin(baudRate, SERIAL_8N1, 14, 12, false);
`

GSM is powered from Li-Ion battery so it should have sufficient power, it's stable.
I tried many different baud rates.

The problem is garbage characters received on serial when GSM module is performing any activity that involves GSM communication packets. So im getting those when: receiving call, making call, regisering to network, establishing tcp connecting etc.

Normally i can send/received AT commands just find:
```
[GSM] => AT
[GSM]  <= OK
[GSM] --- 
[GSM] => AT+CSQ
[GSM]  <= +CSQ: 11,0
[GSM]  <= OK
[GSM] --- 
[GSM] => AT+CBC
[GSM]  <= +CBC: 0,92,4141
[GSM]  <= OK
[GSM] --- 
[GSM] => AT+CPIN?
[GSM]  <= +CPIN: READY
[GSM]  <= OK
[GSM] --- 
[GSM] => AT+CREG?
[GSM]  <= +CREG: 0,5
[GSM]  <= OK
[GSM] --- 
battery: 92%, voltage = 4.14 V
Signal quality: 11, operator = 
```

Hovever dirty garbage starts to show up as soon as I call GSM modem:

```
[GSM] => AT+CSQ
[GSM]  <= +CSQ: 11,0
[GSM]  <= OK
[GSM] --- 
[GSM] => AT+CBC
[GSM]  <= +CBC: 0,91,4132
[GSM]  <= OK
[GSM] --- 
[GSM] => AT+CPIN?
[GSM]  <= +CPIN: READY
[GSM]  <= OK
[GSM]  <= ���������������n�������W����������������o{_��o��������s���o����������V��߿{���������������~������w�������������������߿w�o���������o��������~~����7�
[GSM]  <= RING
[GSM]  <= +CLIP: "+48664500631",145,"",0,"",0
[GSM] --- 
[GSM] => AT+CREG?
[GSM]  <= +CREG: 0,5
[GSM]  <= OK
[GSM] --- 
battery: 91%, voltage = 4.13 V
Signal quality: 11, operator = 
[GSM] => AT
[GSM]  <= }�����������?���������������}�������������~��������������_�����������������������-���v��l��{��w?���׾��������;�����w������O_��=���������w���j��u����?�����
[GSM]  <= OK
[GSM] --- 
```

My guesses are:
1. Interference but how to avoid that? RXD/TXD are quite close to antenna.
2. Something specific to ESP32. Maybe need pullup/pulldown for RX pin on esp32? I already tried gpio_pullup_en gpio_pulldown_en. didn't help?

Any ideas? Help appreciated!

toomasz, 2018-11-07:
Problem solved, I changed RX pin to 16 and ESP doesn't receive garbage on it(unless antenna is very close to it) Seems pin 16 is dedicated for UART so probably have some anti interference/pulldown/whatever built in:
[image]

closed 2019-02-06 by vshymanskyy

mamadgm, 2021-12-30:
I am Still Have The Same Issue :(