vshymanskyy/TinyGSM #210: A6 Module V03.05 in MQTT

cfvelez, 2018-10-14:
Hello, Do you know because the A6 gsm module with version V03.05 doesn't work with protocol MQTT and I get a error -2 MQTT_CONNECT_FAILED the network connection failed. I don't have this problem with the V03.02 version or A7 module and all works fine!. does anybody have idea how I can fix it in v03.05 in A6 gsm module?. I would appreciate any help. Thank you

SRGDamia1, 2018-10-15:
The A6 is nearly identical between those tags.  Can you check which commits work and don't?

cfvelez, 2018-10-15:
The CIPSTART command doestn´t work in V03.05, the response is never received.

SRGDamia1, 2020-04-15:
I'm going to close this older issue.  Please re-open it if you're still having problems.

closed 2020-04-15 by SRGDamia1