vshymanskyy/TinyGSM #212: Did TinyGsm support A9 ?

likewwddaa, 2018-10-17:
A6 was replaced by A9   year ago .
http://wiki.ai-thinker.com/gprs

Did TinyGsm support A9?
Or,How i fixed the source code to supoort A9?

thank u .

eidos85, 2018-12-23:
I'm in the same situation, will this lib support A9 eventually?
Thank you

vshymanskyy, 2019-02-06:
Should work in the same mode as A7.
Currently there are no plans to work on A9

closed 2019-02-06 by vshymanskyy

UksusoFF, 2019-04-26:
Anyone test it with A9G?

rpetteruti, 2019-04-26:
> Anyone test it with A9G?

It doesn't work with A9G, here you can find a version that works with A9G: https://github.com/rpetteruti/A9GTracker

ayazurrehman786, 2022-02-23:
> > Anyone test it with A9G?
> 
> It doesn't work with A9G, here you can find a version that works with A9G: https://github.com/rpetteruti/A9GTracker

Is this work for A9G. Can I use it in ESP32

Mukund2900, 2024-06-13:
> Should work in the same mode as A7. Currently there are no plans to work on A9

It does not work with A7. I tried the following ->
```
#define TINY_GSM_MODEM_A7
#include <TinyGsmClient.h>

// Set your APN
const char apn[]  = "airtelgprs.com";  // type your APN
const char user[] = "";                // Leave empty if not required
const char pass[] = "";                // Leave empty if not required

// Server details
const char server[] = "dummy.restapiexample.com";
const int port = 80; // HTTP port

TinyGsm modem(Serial2);                // Use Serial2 for communication with A9G
TinyGsmClient client(modem);

void setup() {
  Serial.begin(115200);
  delay(500);
  Serial.println("\nInitializing modem...");
  Serial2.begin(115200);               // Initialize Serial2
  delay(2000);

  // Initialize modem using TinyGsm
  Serial.println("Initializing modem using TinyGsm...");
  if (!modem.restart()) {
    Serial.println("Failed to restart modem, attempting to initialize...");
    if (!modem.init()) {
      Serial.println("Failed to initialize modem, please check your connections and reset the board.");
      while (true);
    }
  }

  Serial.println("Modem initialized");

  Serial.println("Connecting to network...");
  if (!modem.waitForNetwork()) {
    Serial.println("Network connection failed");
    while (true);
  }

  Serial.println("Network connected");

  Serial.println("Connecting to GPRS...");
  if (!modem.gprsConnect(apn, user, pass)) {
    Serial.println("GPRS connection failed");
    while (true);
  }

  Serial.println("GPRS connected");
  makeHttpRequest();
}

void loop() {
  // Your code to use the internet connection goes here
}

void makeHttpRequest() {
  Serial.println("Connecting to server...");
  if (!client.connect(server, port)) {
    Serial.println("Connection to server failed");
    return;
  }

  Serial.println("Connected to server, sending GET request...");

  // Make an HTTP GET request
  client.println("GET /api/v1/employees HTTP/1.1");
  client.println("Host: dummy.restapiexample.com");
  client.println("Connection: close");
  client.println();

  // Read all the lines of the reply from server and print them to Serial
  while (client.connected() || client.available()) {
    if (client.available()) {
      String line = client.readStringUntil('\n');
      Serial.println(line);
    }
  }

  // Close the connection
  client.stop();
  Serial.println("HTTP request sent, connection closed");
}

```

But this does not work. 
It is able to restart but network connections are not happening 

Modem initialized
Connecting to network...
Network connection failed

This is what i get.