vshymanskyy/TinyGSM #213: Module being reset when enabling SSL after uploading SSL cert

jameshowe, 2018-10-17:
TinyGSM version: 0.3.5
Hardware: SIM800L, [DeekRobot Pro Micro](https://www.arduino-board.com/boards/dr-pro-micro)

### Scenario, steps to reproduce
1. Flash MC with [SIM800_SslSetCert](https://github.com/vshymanskyy/TinyGSM/blob/master/examples/more/SIM800_SslSetCert/SIM800_SslSetCert.ino) example and upload any of the 3 available SSL certificates to SIM800L module (confirm upload is successful).
2. Flash MC with [HttpsClient example](https://github.com/vshymanskyy/TinyGSM/blob/master/examples/HttpsClient/HttpsClient.ino) and run to completion

### Expected result
The GSM module establishes a TCP to the supplied server in the example (`vsh.pp.ua`) and pulls down the data from the example resource (`/TinyGSM/logo.txt`)

### Actual result
At the point of establishing the TCP connection the SIM800L module is reset thus causing the connection to fail. 

### AT command log
```
Initializing modem...
AT

OK
AT&FZ

OK
ATE0

OK
AT+CPIN?

+CPIN: READY

OK
ATI

SIM800 R14.18

OK
Modem: SIM800 R14.18
AT+CIPSSL=?

+CIPSSL: (0-1)

OK
Waiting for network...AT+CREG?

+CREG: 0,5

OK
 OK
Connecting to <apn>AT+CIPSHUT

SHUT OK
AT+CGATT=0

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","apn"

OK
AT+CGDCONT=1,"IP","apn"

OK
AT+CGACT=1,1
   AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"172.33.116.34"

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

OK
AT+CSTT="mobiledata","",""

OK
AT+CIICR

+SAPBR 1: DEACT

OK
AT+CIFSR;E0

172.33.36.120

OK
AT+CDNSCFG="8.8.8.8","8.8.4.4"

OK
 OK
Performing HTTPS GET request... AT+CIPCLOSE=1

ERROR
AT+CIPSSL=1

OK
AT+CIPSTART=1,"TCP","vsh.pp.ua",443

OK
 failed to connect
```

jameshowe, 2018-10-17:
Oddly, noticed that it only appears to be the initial connection attempt that disconnects the GPRS connection, next time the script loops and reconnects it stays connected but never successfully connects to the server.

tacruc, 2019-05-12:
Did you find a solution. I think I have the same issue. Oddly, it works for me to connect to google.com but github.com and myserver.com fails.

SRGDamia1, 2020-04-15:
I'm going to close this older issue.  Please re-open it if you're still having problems.

closed 2020-04-15 by SRGDamia1